serialport/serialport-rs #37: tcflush on open may cause data loss

jessebraham, 2022-03-08:
This issue was migrated from GitLab. The original issue can be found here:
https://gitlab.com/susurrus/serialport-rs/-/issues/117
---
Currently, serialport-rs issues a `unsafe { tcflush(fd, libc::TCIOFLUSH) };` shortly after opening the serial device.
https://gitlab.com/susurrus/serialport-rs/-/blob/master/src/posix/tty.rs#L123

Opening the device already asserts DTR, so there is a small window where DTR is high but data received in that window will get lost.

This was noticed with an embedded device which starts sending serial data immediately after detecting DTR. It was reproduced on both Linux and MacOS. It probably affects all systems using `posix/tty.rs`.

In this case, the firmware of the embedded device is open source, so inserting a small delay between detecting DTR and sending data worked around the issue. But in general, I think if a host asserts DTR, it should really be ready to receive data.

Unfortunately, it seems to be impossible to flush the buffers before opening the device, and it's also impossible to open the device without asserting DTR. (See for example https://unix.stackexchange.com/questions/446088/how-to-prevent-dtr-on-open-for-cdc-acm)

Therefore, the only solution would be to remove the call to `tcflush` (or make it optional, somehow).

What's the reason for flushing the buffers, is it really necessary?

jannic, 2022-03-10:
Another observation: At its current place, that `TCIOFLUSH` seems to be completely useless, as it happens before the port is configured to its final baud rate (and other settings).
Therefore, there is still a window where garbled data could be received and placed in the input buffer.
My suggestion still is: Just remove that line.

kaluk1321, 2022-03-13:
> Another observation: At its current place, that `TCIOFLUSH` seems to be completely useless, as it happens before the port is configured to its final baud rate (and other settings). Therefore, there [...]

Damn i was just going to ask a question about how to clear the buffer properly since i was still receiving garbage invoking port.clear(ClearBuffer::All) just after opening the port. Is this also useless? Is there a viable way to properly clear the buffer in this version?

jannic, 2022-03-13:
@kaluk1321, that sounds like a different issue. If you call `port.clear(ClearBuffer::All)` manually, that's obviously after `open()` has been executed successfully. So your call should remove garbled bytes received before the settings were applied.

PR #40 (2022-03-22): Remove useless call to tcflush on open
Closes: #37 

This is a draft PR for now, as I didn't have time to test it on real hardware.
closed 2022-03-22 by jessebraham