serialport/serialport-rs #42: Opening Port on Android "Error opening port Permission denied"

glenngartner, 2022-04-05:
I'm using this lib on Android, when when I pass a port name from Java to this lib, I receive "Error opening port Permission denied". Just wondering to what degree Android is supported (I know device enumeration isn't, that's why I'm passing available devices from the Java layer). Some specifics: 

## Android SDK / NDK targets: 
```gradle
    defaultConfig {
        applicationId "..."
        minSdk 27
        targetSdk 30
        versionCode 1
        versionName "1.0"
        ndkVersion = "23.1.7779620"
    }
```
## Android Manifest, giving access to certain devices
```xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="xxx.xxx.xxx">
    <uses-feature android:name =" android.hardware.usb.host"/>
    <application
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:roundIcon="@mipmap/ic_launcher_round"
        android:supportsRtl="true"
        android:theme="@style/xxx">
        <activity
            android:name=".MainActivity"
            android:exported="true"
            android:directBootAware="true"
            >
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
            <intent-filter>
                <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
            </intent-filter>
            <meta-data
                android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
                android:resource="@xml/device_filter" />
        </activity>
    </application>

</manifest>
```
## Device filters, which when connected, prompt the user for permission to access the particular USB device (which, in my case, is a 232 to USB FTDI chip (vendorId: 0403, Product Id: 6001):
```xml
<?xml version="1.0" encoding="utf-8"?>
<resources>
    <!-- 0x0403 / 0x60??: FTDI -->
    <usb-device vendor-id="1027" product-id="24577" /> <!-- 0x6001: FT232R -->
    <usb-device vendor-id="1027" product-id="24592" /> <!-- 0x6010: FT2232H -->
    <usb-device vendor-id="1027" product-id="24593" /> <!-- 0x6011: FT4232H -->
    <usb-device vendor-id="1027" product-id="24596" /> <!-- 0x6014: FT232H -->
    <usb-device vendor-id="1027" product-id="24597" /> <!-- 0x6015: FT230X, FT231X, FT234XD -->

    <!-- 0x10C4 / 0xEA??: Silabs CP210x -->
    <usb-device vendor-id="4292" product-id="60000" /> <!-- 0xea60: CP2102 and other CP210x single port devices -->
    <usb-device vendor-id="4292" product-id="60016" /> <!-- 0xea70: CP2105 -->
    <usb-device vendor-id="4292" product-id="60017" /> <!-- 0xea71: CP2108 -->

    <!-- 0x067B / 0x23?3: Prolific PL2303x -->
    <usb-device vendor-id="1659" product-id="8963" /> <!-- 0x2303: PL2303HX, HXD, TA, ... -->
    <usb-device vendor-id="1659" product-id="9123" /> <!-- 0x23a3: PL2303GC -->
    <usb-device vendor-id="1659" product-id="9139" /> <!-- 0x23b3: PL2303GB -->
    <usb-device vendor-id="1659" product-id="9155" /> <!-- 0x23c3: PL2303GT -->
    <usb-device vendor-id="1659" product-id="9171" /> <!-- 0x23d3: PL2303GL -->
    <usb-device vendor-id="1659" product-id="9187" /> <!-- 0x23e3: PL2303GE -->
[... 64 lines cut ...]
                None
            }
        }
    }
```
## AVC prompts
Sometimes (but not every time) I'll see LogCat print a AVC permissions issue, like this:

```bash
type=1400 audit(0.0:2315): avc: denied { search } for name="usb" dev="tmpfs" ino=608 scontext=u:r:untrusted_app:s0:c236,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=0 app=com.sevenTowers.sensor_android_native
```
## SELinux permissions

I understand, from Google knowledge, that the above comes from the SELinux layer, basically saying I don't have permission to access this device. My USB Service says I have permission to access to the USB, but perhaps the path for the usb is permitted, and the serial protocol requires something different. This is where I'm unsure. 

## My question: 

Does my android device have to be rooted for this library to work? It's not, currently. I'd prefer it not have to be. I know there are java libraries for accessing serial devices on Android specifically (like here https://github.com/mik3y/usb-serial-for-android), because device drivers on stock (unrooted) Android need to be provided on the app, because they're not provided by the OS. I can integrate an entirely different serial lib for Android, and abstract the parsing of each sensor in rust by passing buffers from the above lib into rust, but I'd prefer to use one lib across devices, if possible. 

I've seen the clever folks over at libusb resolve this issue for unrooted Android devices with instructions like this: https://github.com/libusb/libusb/blob/master/android/README (if you're still reading, it begins at line 35)
where they request permission at the Java layer, and pass the fileDescriptor from java to the libusb native code. I'm not that familiar with this level of detail yet, but perhaps there's something to it. Or, perhaps that's just a USB interop thing, not related to serial devices. This is where I can use some help. 

My goal: open a serial port from Android using this lib, parse and handle all the data in rust, as I'm doing with my desktop apps, without having to write too much _more_ interop in Java, so this works on android. 

Thanks!

AlonSpinner, 2024-06-19:
Here for the same issue!

have you made any progress @jessebraham ?

dgramop, 2024-09-11:
My understanding is that serial ports, as we know them from UNIX, are not available to Android apps (at this time). However USB is available if you get permissions and pass the file descriptor down into native code. 

The reason the Kotlin drivers work is because they're really USB Serial drivers implemented in userspace, since the kernelspace drivers expose /dev/ serial ports (apps don't have permissions to these). 

To get access to USB serial ports on an unrooted android, we'd need a rust implementation of USB serial in userspace. This would not give you access to native serial ports on android however, for example if you have some direct serial interface on inside your android phone. 

Enumerating USB devices (and therefore USB serial ports) would still not be possible from Rust, since you must use android's APIs to do this.