serialport/serialport-rs #43: Can't identify virtual com[windows]

duqingkun, 2022-04-15:
It can't identify the virtual com on windows. However, I didn't test on Linux.

mlsvrts, 2022-04-27:
@duqingkun Can you provide the exact error/debug output when you try to open the port? Also, can you share how you are creating the virtual serial ports? (Eltima?)

Currently there isn't enough information here for someone to try and reproduce this issue.

miguelraz, 2024-05-06:
Hello! I have a small reproducer here:

```rust
fn main() {
    let ports = serialport::available_ports().expect("No ports found!");
    for p in ports {
        println!("{}", p.port_name);
    }
    // COM9 was printed for me
    serialport::new("COM9", 9600).open().expect("we done goof'd");
}
```
Will give 
```
❯ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.02s
     Running `target\debug\serialbug.exe`
COM3
COM9
thread 'main' panicked at src\main.rs:7:42:
we done goof'd: Error { kind: NoDevice, description: "The system cannot find the file specified." }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: process didn't exit successfully: `target\debug\serialbug.exe` (exit code: 101)
```
on my Windows machine.