serialport/serialport-rs #52: serde support in the 4.0.x line

fpagliughi, 2022-05-14:
Funny, I was just about to do a PR to add serde support, and saw that it was already added just a few days ago. Awesome!

I was wondering if you folks could possibly cherry-pick that commit (8f657fb67df0f79320ae593316d6bff18364ba87) into the 4.0 line and release it (4.0.2)?

The problem is that we're using Yocto for embedded Linux builds and are stuck with a 2018 Edition of the compiler for now. So we can't move to 4.1 yet. And we'd love serde support.

Thanks

fpagliughi, 2022-05-15:
Alternately reconsider the `version = 2021` requirement.

The library itself seems to compile fine with rustc v1.46.0. The tests seem to use some newer features, going up to around v.154.0 (version 2018).

Considering this library is especially useful in Embedded Linux contexts, it would be really nice to keep the MSRV as old as reasonably possible without making things painful for yourselves. But certainly don't constrain it by artificially specifying a later version than is necessary.

By just changing the version in `Cargo.toml`, it works for me with the latest Yocto:
 ```
edition = "2018"
#rust-version = "1.54"
```

jessebraham, 2022-05-31:
Thanks for this issue! Sorry for taking so long to get around to replying and for breaking compatibility, I didn't realize such relatively old versions of Rust were still commonly used 😁 

I've reverted to the 2018 edition locally and fixed any clippy/other issues, so I will get that pushed and merged. I also plan to publish a new release of this library soon (this or next week) which will include these changes, so you should be able to use this newest version moving forward.

fpagliughi, 2022-06-01:
Nice! Thanks. No worries about the response time; this week I merged a 3 year old PR into one of my project. It was so old that the author had deleted his branch and likely retired from his job.

I think this kind of thing (older compilers) will become more common as Rust becomes more popular and corporate. On this job, we need four different developer groups to approve an update to the compiler!

closed 2022-06-02 by jessebraham