serialport/serialport-rs #56: Migrate from `winapi` to `windows-sys`

jessebraham, 2022-06-21:
The [winapi](https://github.com/retep998/winapi-rs) crate is no longer maintained, and Microsoft are now maintaining the [windows-sys](https://github.com/microsoft/windows-rs) crate (among others), so we should migrate to this.

DoumanAsh, 2022-07-01:
I would recommend not to hurry with it because `windows-sys` makes semver incompatible bumps with each new version so it would become annoying and eventually stale in this crate

caspermeijn, 2024-08-30:
Today I tried this conversion, but it is currently not straightforward. 

The DCB struct has a lot of bitfields, which are much used in `serialport`. The bitfields don't have easy getters/setters yet: https://github.com/microsoft/windows-rs/issues/2942 Therefore doing the conversion today would decrease the code readability.

sirhcel, 2024-09-26:
Thank you for attempting the migration @caspermeijn! Do you have some work-in-progress to share with us (at best in a fork) so that I can have a look at how exactly this gets messy?

caspermeijn, 2024-10-02:
> Thank you for attempting the migration @caspermeijn! Do you have some work-in-progress to share with us (at best in a fork) so that I can have a look at how exactly this gets messy?

This was my work in progress commit: https://github.com/caspermeijn/serialport-rs/commit/58ab9ba04293ba692e263f615f4b4bf45dbe0bcf

I didn't work on the bit fields when I saw the mentioned lack of support. The problem comes down to this:
[`DCB::_bitfield`](https://docs.rs/windows-sys/0.59.0/windows_sys/Win32/Devices/Communication/struct.DCB.html#structfield._bitfield) is an integer field, but it represents multiple bit fields. The `windows-rs` crate doesn't provide any help, so `serialport` needs to maintain a list of the bit indexes. That sounded like not the core competency of this crate and should be provided by a Windows API crate instead. The maintainer of `windows-rs` seems to be aware of that.