serialport/serialport-rs #57: 'list_ports.rs' example doesn't work

birdistheword96, 2022-06-30:
The list ports example doesn't work on windows, as its missing the 'interface' field from the struct:

```
error[E0609]: no field `interface` on type `UsbPortInfo`
  --> src\main.rs:33:34
   |
33 | ...                   info.interface
   |                            ^^^^^^^^^ unknown field
   |
   = note: available fields are: `vid`, `pid`, `serial_number`, `manufacturer`, `product`
```

Also with that section commented out, no useful information is printed out. I have 4 bluetooth ports (actually just 2 bluetooth devices but windows makes 2 ports for each) and 2 virtual com ports,  but the 'type' for all of them is just 'Unknown':

```
Found 6 ports:
  COM7
    Type: Unknown
  COM6
    Type: Unknown
  COM8
  COM9
    Type: Unknown
  COM10
    Type: Unknown
  COM11
    Type: Unknown
    
```

Using **serialport-rs v4.2.0** with **rust 1.61.0 (stable)** on Windows 10

mlsvrts, 2022-07-01:
I think the `list_ports` from the head of this repository will only be compatible with the current development release (4.2.1-alpha). I expect the example at the 4.2.0 release (https://github.com/serialport/serialport-rs/commit/60d1f3eb1d455ef44553d54b97c1cd4b821b45fb) will run without any errors. 

However I don't know that the reported port information for non-usb ports will be any more detailed.

sirhcel, 2023-05-22:
[4.2.1](https://crates.io/crates/serialport/4.2.1) got released and is available on crates.io. This release comes with the feature `usbportinfo-interface` to consistently build the library and its examples. Building this example from the repository for windows works without
```
$ cargo build --target x86_64-pc-windows-gnu --example list_ports
```
or with this feature
```
$ cargo build --target x86_64-pc-windows-gnu --features usbportinfo-interface --example list_ports
```
works for me.

Please feel free to reopen this issue if you still encounter problems with 4.2.1.

closed 2023-05-22 by sirhcel