serialport/serialport-rs #62: On the USB device running Windows, the product name of the non-UTF-8 device is garbled

ma6254, 2022-08-18:
- OS: win10(Code page 936)

```rust
    let serial_list = serialport::available_ports();

    match serial_list {
        Ok(serial_list) => {
            println!("{:?}", serial_list.len());
            for s in serial_list {
                println!(
                    "{} {}",
                    s.port_name,
                    match s.port_type {
                        // serialport::SerialPortType::UsbPort(usb_info) => usb_info.product.unwrap(),
                        serialport::SerialPortType::UsbPort(usb_info) => {
                            format!("{:?}", usb_info.product.unwrap())
                        }
                        serialport::SerialPortType::PciPort => String::from("PciPort"),
                        serialport::SerialPortType::BluetoothPort => String::from("BluetoothPort"),
                        serialport::SerialPortType::Unknown => String::from("Unknown"),
                    },
                );
            }
        }
        Err(e) => {
            println!("{:?}", e);
            return;
        }
    }
```

```
4
COM27 "USB �����豸 (COM27)"
COM11 "USB �����豸 (COM11)"
COM12 "JLink CDC UART Port (COM12)"
COM10 "USB-SERIAL CH340 (COM10)"
```

[image]

mlsvrts, 2022-08-23:
@ma6254 Thanks for reporting this; it looks like we're using the ANSI version of the windows API, which is probably why this breaks. Can you try this patched branch and let me know if it fixes your issue?

https://github.com/mlsvrts/serialport-rs/tree/fix-unicode-names

ma6254, 2022-08-24:
`commit id : 9f65b84c04c2c002f20371f345ec8027751542c0`
it not work
It's frustrating. I think the problem is unsolvable
Hardware designers consider only the local language and use a specific encoding format, such as "GBK", rather than the standard "UTF-8"
so sad 😥

mlsvrts, 2022-08-25:
@ma6254 Actually, this may have been an issue with my patch! I checked again, and the device manufacturer and "friendly name" are read from the registry keys -- this was still being done with an ANSI windows API. 

I have swapped that with the unicode version on the above branch (see commit: https://github.com/serialport/serialport-rs/commit/b53f5f3369912b0fbc66d0cfb209ecaaac636191)

Possibly we could have issues with the USB descriptor, but I think it's very likely that the registry strings are proper unicode.

ma6254, 2022-08-25:
@mlsvrts 
is works
```
> cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
     Running `target\debug\aaa.exe`
Hello, world!
1
COM10 USB 串行设备 (COM10)
```

PR #63 (2022-11-30): Fix unicode names
Switch to unicode Windows APIs:

The default use of ANSI Windows APIs prevents some device names from displaying correctly; this can be resolved by using the Unicode API calls. Closes #62
closed 2022-11-30 by sirhcel