serialport/serialport-rs #64: BufRead::read_line() doesn't work when timeout is set to 0

Bktero, 2022-10-11:
Hello

I have a code like this:

```rust
let port = serialport::new(port_name, baud_rate)
    .timeout(Duration::from_millis(0)) // ----- NO TIMEOUT HERE
    .open()
    .expect("Error: ");

let mut port = BufReader::new(port);
let mut line_buffer = String::new();

match port.read_line(&mut line_buffer) {
    Ok(size) => println!("{} bytes read -> {:?}", size, line_buffer),
    Err(e) => eprintln!("Error while reading: {:?}", e),
}
```

It will never receive a line. Change the timeout to any value and the line will be received (apparently, when the timeout expires).

This behavior doesn't seem intuitive to me.

starovoid, 2023-02-01:
I believe it is necessary to avoid zero timeouts, perhaps it is worth adding a corresponding 'assert' into the '.timeout()'.

DanielJoyce, 2023-05-05:
Timeout means it immediately returns, and will never read a line.