serialport/serialport-rs #66: Enumate the serialport

zrz4066, 2022-10-19:
pi@raspberrypi:~ $ sudo ./rpidemo
Blinking an LED on a Raspberry Pi Compute Module 4.
/sys/class/tty/ttyUSB3
/sys/class/tty/ttyUSB1
/sys/class/tty/ttyUSB2
/sys/class/tty/ttyUSB0

sirhcel, 2023-07-04:
I'm sorry for the late reply @zrz4066! Which release of serialport-rs were you using back then? Could you please give our example [`list_ports`](https://github.com/serialport/serialport-rs/blob/v4.2.1/examples/list_ports.rs) a try? On a Raspberry Pi 3+ running Raspian Buster, it lists me the expected USB serial devices:
```
$ cargo run --example list_ports
[...]
Found 3 ports:
  /dev/ttyAMA0
    Type: Unknown
  /dev/ttyUSB1
    Type: USB
    VID:10c4 PID:ea60
     Serial Number: [...]
      Manufacturer: Cygnal Integrated Products, Inc.
           Product: CP2102/CP2109 UART Bridge Controller [CP210x family]
  /dev/ttyUSB0
    Type: USB
    VID:0403 PID:6001
     Serial Number: FT[...]
      Manufacturer: Future Technology Devices International, Ltd
           Product: FT232 Serial (UART) IC
```
Which operating system release are you running on your Raspberry Pi? If the output differs there, then this is something to look further into.

I've got the output above even with our release 4.2.0 which was out at the time you were writing this issue.

sirhcel, 2023-08-08:
Finally managed to fabricate the output from the description with the `list_ports.rs` example and disabled support for libudev:
```
$ cargo run --no-default-features --example list_ports
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
     Running `target/debug/examples/list_ports`
Found 5 ports:
  /sys/class/tty/ttyUSB1
    Type: Unknown
  /sys/class/tty/ttyS0
    Type: Unknown
[...]
```

It looks like the actual enumeration just returns directory names from `/sys/class/tty`. These are not the device file paths needed for instantiating the devices like `/dev/ttyUSB0`. What likely needs to be done:
* Check if there is a symlink pointing to the actual device file from inside `/sys/class/tty/ttyX`?
* Or just build the device file name `/dev/ttyX` from `/sys/class/tty/ttyX`
    * Cross check that the device numbers from `/sys/class/tty/ttyX/dev` match?

pandakka, 2023-08-10:
Version 4.2.2
 Exec code: example list_ports   (no-default-features)
 Result  :    No ports found.      （actually have ttymxc1 ttymxc2）
 Hardware：NXP.I.MX6ULL   
 Kernel  version：4.5.1 
 Build Env :  --target=arm-unknown-linux-musleabihf

Thanks for your Answer  at #119  and hope it can help you.

sirhcel, 2023-08-10:
> Result : No ports found. （actually have ttymxc1 ttymxc2)

I would have expected them to be listed in `/sys/class/tty` @pandakka. How does for example `/sys/class/tty/ttymxc1` look like? Could you please capture a directory listing (`ls -l`) there? I'm wondering why [the enumeration](https://github.com/serialport/serialport-rs/blob/22f854dbd43966abb6be216db686a06074ccdef1/src/posix/enumerate.rs#L475) fails.

pandakka, 2023-08-11:
@sirhcel Sure  ! But I need to  go outside these days.Once back I will write in this issues.

sirhcel, 2023-08-23:
I found the time to boot up an i.MX8 board. It runs a Linux with BusyBox and in `/sys/class/tty`, the links for `ttymcx0` and `ttymxc2` are present and the linked directories look like they would result in these devices being enumerated'

```
# uname -a
Linux phyboard-polis-imx8mm-3 4.19.35-bsp-yocto-fsl-i.mx8mm-pd20.1.0 #1 SMP PREEMPT Tue Jun 30 01:59:19 UTC 2020 aarch64 GNU/Linux
# ls -l /sys/class/tty/ttymxc*
lrwxrwxrwx    1 root     root             0 Aug 23 22:11 /sys/class/tty/ttymxc0 -> ../../devices/platform/30860000.serial/tty/ttymxc0/
lrwxrwxrwx    1 root     root             0 Aug 23 22:11 /sys/class/tty/ttymxc2 -> ../../devices/platform/30880000.serial/tty/ttymxc2/
# ls -l /sys/class/tty/ttymxc0/device /sys/class/tty/ttymxc2/device
lrwxrwxrwx    1 root     root             0 Aug 23 22:14 /sys/class/tty/ttymxc0/device -> ../../../30860000.serial/
lrwxrwxrwx    1 root     root             0 Aug 23 22:16 /sys/class/tty/ttymxc2/device -> ../../../30880000.serial/
```

Linux on this board is pretty bare and getting `list_ports` to build looks like a pretty bunch of work ahead. So I'm stopping here for now.

How does things look on your system @pandakka?

pandakka, 2023-09-28:
Sorry for the late reply!  It seems looks like same  as mine.  Due to the system is pretty bare, that  libudev  may not work which cause the  list_ports  can't find the port. Right?  Thank you for you answar！