serialport/serialport-rs #70: New release?

tmbull, 2022-11-27:
Hi, thanks for excellent library. I was just wondering if there is any reason you all couldn't cut a new release of the library? It looks like the last release was back in June, and it would be nice to have some of the fixes that have been merged in since then in a release version. (I am specifically interested in #58 😄 )

jessebraham, 2022-11-29:
There is no reason we cannot cut a new release, I've just been busy 😁 There are a couple open PRs I'd like to get merged first, so I'll try to get those taken care of and then I will publish a new release as soon as possible!

tmbull, 2022-12-01:
Understood. Thanks for the reply. I would offer to help maintain, but I think I am too new to Rust to be comfortable maintaining a low-level library like this. I'll close this issue as you've answered my question.

closed 2022-12-01 by tmbull