serialport/serialport-rs #73: bufReader read_until not handled correctly on Windows

craigkai, 2022-12-06:
Using the following code:

```rust
 let mut reader = BufReader::new(connection);
 let mut my_str = vec![];

if let Err(error) = reader.read_until(0xA, &mut my_str) {....}
```

Where I am reading some data from a serial connection. On Linux & Mac I see the read_until function executing quite fast, but on Windows (11) I see the serial connection's timeout (500ms) is what is ending the read_until command -- Which does then proceed to do the correct thing after with the correct value in the buffer. It is just comparatively slow.

An example of the output I am seeing:
```
Successfully sent write to serial: CONFIG
Successfully read from serial "0\n"
Successfully sent write to serial: ILLUMDAY
Successfully read from serial "10\n"
Successfully sent write to serial: ILLUMNIGHT
Successfully read from serial "25\n"
Successfully sent write to serial: ILLUMMODE
Successfully read from serial "0\n"
Successfully sent write to serial: ECHO
Successfully read from serial "0\n"
Successfully sent write to serial: ANIM
Successfully read from serial "0\n"
```

Is this some kind of bug with Windows or am I missing something obvious? Thanks for any input!

mlsvrts, 2022-12-07:
Fast `BufReader` example:

```rust
use std::io::{BufReader, BufRead};

fn main() {
    let mut port = serialport::new("COM3", 9600)
        .timeout(std::time::Duration::from_millis(500))
        .open()
        .expect("failed to open port");

    let reader = port.try_clone()
        .expect("failed to clone port");

    port.write_all(&[0xDE, 0xAD, 0xBE, 0xEF, 0x00])
        .expect("failed to write to port");

    // Create the slow bufreader
    let mut buf = vec![];
    let mut b_reader = BufReader::with_capacity(100, reader);

    let now = std::time::Instant::now();
    if let Err(e) = b_reader.read_until(0x00, &mut buf) {
        println!("Slow reader error: {:?}", e);
    }

    println!("slow read time: {}ms", now.elapsed().as_millis());

    // Fast bufreader
    port.write_all(&[0xDE, 0xAD, 0xBE, 0xEF, 0x00])
        .expect("failed to write to port");
    
    let mut b_reader = BufReader::with_capacity(1, b_reader.into_inner());

    let now = std::time::Instant::now();
    if let Err(e) = b_reader.read_until(0x00, &mut buf) {
        println!("Fast reader error: {:?}", e);
    }

    println!("fast read time: {}ms", now.elapsed().as_millis());
}
```

Which gives me:

```
slow read time: 515ms
fast read time: 19ms
```

craigkai, 2022-12-07:
This has been so helpful, thank you so much for all the assistance!!

DanielJoyce, 2022-12-18:
To me this perhaps implies we need to refactor things so this is more apparent to the user. Windows behavior is apparently commonly counterintuitive.

That or the default windows serial port config needs to be changed to better mirror POSIX behavior and this information added to the docs.

larsch, 2023-02-01:
I have suggested a change to use the POSIX-like blocking read (with early return on ANY data) and timeout in #79. This fixes the general case of of read blocking until the buffer is full. It only makes it slightly more CPU intensive as it will return as much as in the FIFO any time the read call is invoked, but for serial port speeds, this is irrelevant.

This change is necessary for any serial port protocol that uses request/response scheme where the size of packets is unknown in advance, or when using mio-seriel/tokio-serial where you can't tell the underlying library how much data you want to read.

jerrywrice, 2023-04-15:
I'm just starting to use serialport crate, but would it be possible to incorporate a new (Windows specific) feature using new functions and other necessary extensions that allows a user (developer) to explicit enable the POSIX-like read behavior larsch describes? This would seem to be a non-breaking change, since it would be only 'new' user code that would activate this feature.