serialport/serialport-rs #85: Dead links in platforms.md

atouchet, 2023-03-18:
https://github.com/serialport/serialport-rs/blob/d60df349623f9ae526513a12bee5b756d7c769bd/doc/platforms.md?plain=1#L27-L28

sirhcel, 2023-03-18:
Thank you very much for pointing this out! I have no experience with NetBSD, but since you are here: are the following links the man pages you were looking for:

- https://man.netbsd.org/NetBSD-8.0/i386/tty.4
- https://man.netbsd.org/NetBSD-8.0/i386/com.4

Do you see anything against pointing to the man pages for the current NetBSD release and without a specific archtiecture? Like:

- https://man.netbsd.org/tty.4
- https://man.netbsd.org/com.4

eldruin, 2023-03-20:
@atouchet would you mind opening a PR fixing the links?

atouchet, 2023-03-21:
Sure, I opened a PR here: #86

PR #86 (2023-03-21): Update NetBSD man page links
Update links to working versions and point to non-specific architecture pages as suggested by @sirhcel.

Fixes #85.
closed 2023-03-21 by eldruin