serialport/serialport-rs #90: After dropping lock serialport is still locked

craigkai, 2023-04-02:
```
pub async fn drop_connection<R: Runtime>(
    serial_state: State<'_, SerialState>,
    window: Window<R>,
) -> Result<String, SerialError> {
    let guard = serial_state.connection.lock().await;
    drop(guard);
}
```
I have the following code where I have  a serial connection being made of type
```
pub struct SerialState {
    pub port: Arc<Mutex<String>>,
    pub connection: Arc<Mutex<Option<Box<dyn serialport::SerialPort>>>>,
    pub baud_rate: u32,
    pub ports: Arc<Mutex<Vec<SerialPort>>>,
}
```
Everything works well on Mac but on Windows after dropping my connection and trying to re-connect I am seeing:
```
'COM4': Access is denied.
```
It looks like the port is still locked by my program. Are there any suggestions on where I am going wrong? Thanks!

mlsvrts, 2023-04-05:
Actually I think the issue might be that Mac is never locking the port at all: 

Looking at your 'drop connection' function, I'm pretty sure it's only dropping the mutex guard (which releases the mutex lock, but doesn't drop the held item)

DanielJoyce, 2023-05-10:
```
pub async fn drop_connection<R: Runtime>(
    serial_state: State<'_, SerialState>,
    window: Window<R>,
) -> Result<String, SerialError> {
    let guard = serial_state.connection.lock().await;
    drop(guard); // Not needed, guard is dropped at end of function anyways. 
}
```

Dropping the guard only drops the lock on the mutex, it does not drop the port.

The logic of this function should be in a Drop impl for SerialState

sirhcel, 2023-08-08:
Did you try the solution proposed in https://github.com/serialport/serialport-rs/issues/90#issuecomment-1542933237 @craigkai? Did it drop/close the port?

craigkai, 2023-08-08:
Its been a bit but Daniel's comment about mearly dropping the lock was correct, I used `take()` on the guard and then dropped that

sirhcel, 2023-08-08:
Thanks for your reply! Then it looks like there is nothing left do do here.

closed 2023-08-08 by sirhcel