serialport/serialport-rs #92: New release?

g0hl1n, 2023-04-05:
Hi @eldruin,
thanks for stepping up as maintainer.

Would you mind tagging a new release for serialport-rs?
My project is waiting for a tag containing the new nix version :wink: 

Thanks & regards

eldruin, 2023-04-05:
Sure, however, it seems no changes have been noted in the changelog since last release.
Would you mind going through PRs merged since then and adding short notes about them in the changelog?

mlsvrts, 2023-04-05:
Just a note that we have at least one (and sort-of two) breaking changes currently in the main branch: 

- The updated `nix` version is technically breaking, if unlikely to cause problems 
- There's a new member in a public struct (`UsbPortInfo.interface`)

We could target the master branch as 5.0.0-alpha and back port a 4.X release without breaking changes. 

Note that we could probably feature-gate the `UsbPortInfo` change to make it non-breaking, but I don't know if we have an easy solution for the `nix` update...

mlsvrts, 2023-04-05:
This is per @jannic (thanks!):

That's a limitation of semver-checks. See https://github.com/obi1kenobi/cargo-semver-checks/issues/5, bullet point "upgrading to new major version of dependency while exporting a type that implements a trait from the dependency".

And yes, I think we are exporting such a type. `impl From<nix::Error> for Error` in `error.rs`.

Not literally a "trait from the dependency", but a trait which includes nix in its generic type.

I guess this won't cause any actual breakage in practice, though: I don't see any reason why external code should try to convert a `nix::Error` into a `serialport::Error`. This code is probably only used from inside the serialport crate.

This code works if the version of `nix` is identical:

```
fn main() {
    let nix_error: nix::Error = nix::Error::UnknownErrno;
    let serialport_error: serialport::Error = nix_error.into();
    println!("{serialport_error:?}");
}
```

However, if the binary crate and serialport use different versions of nix, compiling fails:

```
error[E0277]: the trait bound `serialport::Error: From<Errno>` is not satisfied
 --> src/main.rs:3:57
  |
3 |     let serialport_error: serialport::Error = nix_error.into();
  |                                                         ^^^^ the trait `From<Errno>` is not implemented for `serialport::Error`
  |
  = help: the following other types implement trait `From<T>`:
            <serialport::Error as From<libudev::error::Error>>
            <serialport::Error as From<nix::errno::consts::Errno>>
            <serialport::Error as From<std::io::Error>>
  = note: required for `Errno` to implement `Into<serialport::Error>`
```

So I'm quite sure this technically is a breaking change, but in practice it could be ignored.

eldruin, 2023-04-05:
Interesting! Thank you for looking into it.

sirhcel, 2023-05-10:
What about a maintenance release with simply omitting the breaking changes to `UsbPortInfo` and ignoring the issue with `nix` as proposed in https://github.com/serialport/serialport-rs/issues/92#issuecomment-1497712139?

I'm starting a release branch with #100.

sirhcel, 2023-05-21:
Inching forward with #103.