serialport/serialport-rs #95: Hardware flow control permanently asserts RTS

xioTechnologies, 2023-04-26:
RTS is permanently asserted regardless of `FlowControl::None` or `FlowControl::Hardware`.  Hardware flow control using RTS/CTS signals should assert the RTS output when the receive buffer is full, and inhibit transmission when the CTS input is asserted.  I have confirmed that transmission is inhibited by the CTS input.  The issue is limited to the RTS behaviour.

**This is an urgent issue for us.**  If a fix is not easily achievable then I suggest a workaround where RTS is permanently unasserted.  This will at least allow reception of data from devices using hardware flow control, even if there is a risk that the receive buffer overflows.

A potentially related issue is that I cannot find any scenario where `write_request_to_send` is able to write the RTS level.

I am testing on Windows 10 using an [FTDI USB to UART](https://www.sparkfun.com/products/9873).  This issue was first reported as [#118](https://gitlab.com/susurrus/serialport-rs/-/issues/118#note_880131292) on GitLab.

eldruin, 2023-04-27:
Please feel free to make a change proposal as a pull request. Then the community can discuss the concrete approach.

xioTechnologies, 2023-05-12:
I have misreported this issue.  We were mistakenly measuring DTR not RTS.  I have confirm that the DTR pin state can be written using `write_data_terminal_ready`.

closed 2023-05-12 by xioTechnologies