serialport/serialport-rs #96: how to open a COM port with serialport::new in Windows

kinire98, 2023-04-28:
I'm doing a program to detect the kind of devices that are connected to different COM ports. But with serialport::new in windows I can't seem to find the path to the device that the function asks me to give in Windows. I know that in Linux is /dev but i can't find something similar in Windows. The only thing that i found it's the path to a Windows object or the path of a registry key that can't seem to find. I've been stuck several days with this. Using WSL is not an option. I've seen that many people here have been able to open the port, so if someone can help me I'd be very thankful

kinire98, 2023-04-28:
I just found it. For anyone wondering or that might visit in the future the path is "\\\\.\\COM1" (or the name of another COM). The real path is "\\.\COM1", but you have to add another backslash, like any other path in windows. Hope this helps someone.

closed 2023-04-28 by kinire98

zszep, 2024-01-23:
On my system it is "\\.\\COM1". Anyway thanks for the tip, I tried "COM1" but that would not work.