serialport/serialport-rs #97: Read not reading all the data off a TTY port

Austin-Crabtree, 2023-05-04:
Using: 
- Beaglebone Black Rev C
- Linux Debian 10
- Rust 1.68.2
- Serialport 4.2.0

when reading off a Beaglebone Black UART connection I am running into an issue when I have a message larger than 52 bytes not being pulled off the TTY buffer in one read, typically takes two reads. I am creating a sufficiently large enough buffer and the read isn't filling the vector. I have tried both  `open()` and `open_native()` on the port with no help. I also have narrowed it down to the read being the issue becuase I have put a `clear(ClearBuffer::All)` after reading and parsing the data and I can't get the second half of the data anymore.

Austin-Crabtree, 2023-05-04:
I haven't check the bytes_to_read function I can do that. But the amount of bytes to read is is variable based on the type of message that is being sent. All other messages are requests and don't have a data payload so each other message is the same size about 14 bytes in size.

Austin-Crabtree, 2023-05-04:
I will also say that I checked against my rust application by creating a quick python script using pyserial and was able to read all the data in one read.

DanielJoyce, 2023-05-05:
Raspberry PI Uart is 16 bytes. PySerial probably caches it to another buffer transparently to read by other processes.

DanielJoyce, 2023-05-05:
https://github.com/pyserial/pyserial/blob/31fa4807d73ed4eb9891a88a15817b439c4eea2d/serial/serialposix.py#L570

Here, it reads into a buffer in a loop, only returning when the buffer is full, or timeout.

You will need to imple the same logic as serialport-rs is pretty low level.

newcomertv, 2023-07-21:
This seems like a bug not a feature.. I'd expect a serial interface reader to read until the buffer is full or the transmission has timed-out if I can configure buffer size and timeout duration..

Is there a use-case where this is desired behaviour?

DanielJoyce, 2024-04-10:
Its a feature since folks may want to buffer, some don't. Since you want to buffer, you need to buffer yourself.