serialport/serialport-rs #102: Missing serial_number

Massimiliano-solutiontech, 2023-05-18:
On macOS I can't get the `serial_number` of a device, I know that a serial is present because I programmed it using the `CP210xSetIDs` from SiliconLabs and for test I used `rusb` and `libusb` and I can see the serial number

```toml
serialport = "4.2.0"
```

```rust
&device = UsbPortInfo {
    vid: 4292,
    pid: 60000,
    serial_number: None,
    manufacturer: Some(
        "Silicon Labs",
    ),
    product: Some(
        "Test device",
    ),
}
```

mlsvrts, 2023-05-18:
I believe this was fixed by 

https://github.com/serialport/serialport-rs/pull/65

Can you see if the issue is resolved if you use the head of this repository instead of that release?

Massimiliano-solutiontech, 2023-05-18:
Yes I think so. How can I install the patched version?

Massimiliano-solutiontech, 2023-05-18:
I did this

```toml
serialport = { git = "https://github.com/serialport/serialport-rs", branch = "main" }
```

how soon will be available a new stable version with that patch?

sirhcel, 2023-05-18:
We are working on releasing a maintenance release to 4.2.0 with #100 or #101. You could follow these issues for staying up to date.

closed 2023-05-22 by Massimiliano-solutiontech