serialport/serialport-rs #104: Encountered a problem during the process of making this library into a Flutter plugin

fuyuku, 2023-05-28:
I generated a global reference using Box:: leak so that it can be used in Flutter, but now I cannot close the port. 

Is there any way to close the port?

sirhcel, 2023-05-29:
No, there is no explicit way of closing a serial port besides dropping it.

Could you please elaborate your usage a bit more in depth? Is leaking really required in your use case? In case it is and you're at the point where you can guarantee that it won't be used any further, wouldn't [`Box::from_raw`](https://doc.rust-lang.org/std/boxed/struct.Box.html#method.leak) allow to reconstruct the boxed serial port and drop it?

fuyuku, 2023-05-30:
Actually, I just want a global reference so that I can use it in every function of ffi in the plugin. At present, I can only find Box:: leak.

I tried Box:: into_raw, it seems that it won't work.

If you have any good suggestions, please let me know.

sirhcel, 2023-05-30:
> Actually, I just want a global reference so that I can use it in every function of ffi in the plugin. At present, I can only find Box:: leak.
> 
> I tried Box:: into_raw, it seems that it won't work.

To my understanding, `Box::into_raw` is the way and this works in [my example](https://github.com/sirhcel/serialport-rs/blob/issue-104-box-from-raw/examples/close_from_raw.rs)
```
$ cargo run --example close_from_raw -- --port /dev/tty.usbserial-0
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `/[...]/serialport-rs/target/debug/examples/close_from_raw --port /dev/tty.usbserial-0`
Opened serial port SerialPort ( name: /dev/tty.usbserial-0 baud_rate: 9600 data_bits: Eight flow_control: None parity: None stop_bits: One )
Converted port into raw pointer 0x600000d18000
Converted raw pointer back to port SerialPort ( name: /dev/tty.usbserial-0 baud_rate: 9600 data_bits: Eight flow_control: None parity: None stop_bits: One )
TTYPort::drop
posix::tty:close(3)
Dropped (and closed) port.
```

> If you have any good suggestions, please let me know.

Without some more details and example code, I'm running out of ideas for now.

fuyuku, 2023-05-30:
`pub extern "C" fn init_serial()->*mut dyn SerialPort{
    let port = serialport::new("COM1", 115200).timeout(Duration::from_millis(50)).open().expect("error");
    println!("success");
    Box::into_raw(port)
}
`
When I generated a file using cbindgen, an error occurred and the type is not supported.

sirhcel, 2023-05-30:
Alright, this sheds some more light onto the issue. Attempting, to compile the snipped above, the Rust compiler already points out that it refuses to pass trait objects (which `*mut dyn SerialPort` is) across FFI boundaries. So `cbindgen` being grumpy about this type seems just fine to me.

Following the fundamental theorem of software engineering and wrapping the trait object into a FFI compatible struct pleases both, the Rust compiler and `cbindgen`:
```rust
use serialport::SerialPort;
use std::time::Duration;

#[repr(C)]
pub struct PortHandle {
    port: Box<dyn SerialPort>,
}

#[no_mangle]
pub extern "C" fn open_port() -> *mut PortHandle {
    let port = serialport::new("/dev/tty.usbserial-2104202001", 115200)
        .timeout(Duration::from_millis(50))
        .open()
        .expect("error");
    let handle = Box::new(PortHandle { port });
    Box::into_raw(handle)
}

/// # Safety
///
/// It is safe to call this function at most one time for a port handle obtained via [open_port].
#[no_mangle]
pub unsafe extern "C" fn close_port(handle: *mut PortHandle) {
    if !handle.is_null() {
        let cooked = Box::from_raw(handle);
        drop(cooked);
    }
}
```
For the code above
```
$ cbindgen --lang c
```
produces the following header
```cpp
#include <cstdarg>
#include <cstdint>
#include <cstdlib>
#include <ostream>
#include <new>

struct PortHandle;

extern "C" {

PortHandle *open_port();

/// # Safety
///
/// It is safe to call this function at most one time for a port handle obtained via [open_port].
void close_port(PortHandle *handle);

} // extern "C"
```
which allows me to open and close the port from a C binary.

Now I'm wondering who does your binding for Flutter handle the actual I/O on a serial port?

fuyuku, 2023-05-31:
After testing, you have solved my problem.

> Now I'm wondering who does your binding for Flutter handle the actual I/O on a serial?

I'm a bit confused about this question.

According to my understanding, serialport-rs process the actual I/O and then return the data to Flutter.

sirhcel, 2023-05-31:
> After testing, you have solved my problem.

I'm glad to hear that.

> I'm a bit confused about this question.
> 
> According to my understanding, serialport-rs process the actual I/O and then return the data to Flutter.

Sorry for the confusion. To my understanding the type of the handle to pass from the FFI shim to the Flutter application is the same for I/O operations and closing the port. Now I'm wondering which port handle type you used in the first place for I/O operations.

closed 2023-05-31 by sirhcel

fuyuku, 2023-06-02:
When I use Box::leak, handle is not passed, so I can use flutter_ rust_ bridge_codegen very convenient for generating code.
After I use Box::into_raw, I can't use it anymore