serialport/serialport-rs #106: Bootstrap (or resurrect) a cross-platform port-to-port serial transmit/receive test application

jerrywrice, 2023-06-08:
I'll review existing examples and formulate/generate a two port (cross connected) send/receive test application for cross-platform usage. Initial goal is to allow assessment of existing functional behavior and timing, in order to assist in planning issue fix priorities and future evolution of the library. Goal is to have something to share early the week of June 12th.

sirhcel, 2023-06-08:
I worked on [`examples/hardware_check.rs`](https://github.com/sirhcel/serialport-rs/commit/379930975f7e587d684f3ba7ad978ecc78e48f22) a while ago but stalled due to lack of time.

jerrywrice, 2023-06-14:
Sirhcel,

Based on your feedback, I understand that the immediate plan for crate 'serialport-rs' is to cleanup existing examples and library logic so they function robustly, but without significantly modifying existing example functionality.

With that clear, I'll generate a new example application (named 'bulk_xfr.rs)' which transmits a specified quantity of sequenced serial data. 

The transmitted data will be an auto-generated byte stream consisting of an initial (leading) ASCII decimal transfer-length string field which is space/blank (0x20) terminated, followed by an auto-generated sequence of ascending binary bytes cycling between 0 and 255 in a repeated fashion that satisfies the user designated transfer length. The receiving logic uses the leading ASCII decimal length field string value to derive its actual expected receive length.

Each port's flow control setting will default to 'None'.

There will be command line arguments which specify: 
1. The transfer port name (--PORT="/dev/ttyUSB1"), 
2. The current role, either (--role="XMT") or (--role="RCV"), 
4. The number of sequenced value bytes for the transmit (--XFRLEN="nnnn"), 
5. The baud rate (--BAUD="250_000"). 

This new example may be used between two ports on the same (platform) computer system, or between two computer systems running different OSes. 

Once I verify this new example with a connected Windows and Linux test system while undertaking a lengthy transfer at a commonly supported high baud rate (such as 250K), I will then request someone with a Mac to assist by verifying it as well.

Additional functionality, such as timeout testing and flow-control. could be added to this or another new example once the core ''bulk_xfr.rs' example is working as described above.

Does this sound reasonable - your thoughts?

sirhcel, 2023-06-15:
Further reading: We had some serious amount of discussion in #107

jerrywrice, 2023-07-17:
I've generated and tested a serialport crate stand-alone test and characterization application, named 'receive_timing_info', which characterizes the write() and read() trait methods timing under various operating conditions. I've built and tested it on both Windows 10 and Ubuntu 22.04 LTS, using two FTDI USB serial port cables. The test currently accepts ten (10) command line arguments which allows multiple testing scenarios. I've written a corresponding support document "SerialPort_rs Receive_Timing_Info Test Description.pdf" which details this. I've also generated and tested, the test application's source file, named 'receive_timing_info.rs'. The document describes building the test application from the serialport-rs crate's examples folder with cargo build. This test application saves all of its run-time metrics data in a log-file, since there is impractical to display it all on the console (and this would significantly slow down the run-time execution. Finally, since developing this test I've used it to characterize the Windows write() and read() behavior, especially its blocking and time-out characteristics, and have identified some clear differences between the Windows and Linux crate behavior, which is described in the document. I've also generated and tested a simple patch to the crates' Windows version of 'set_timeout()' which eliminates a fairly important problem with the current windows timeout logic. I would like to cover this (in summary) at the planned August 17 crate planning meeting.