serialport/serialport-rs #112: Dependency on unmaintained crates

newAM, 2023-07-24:
This is minor, but I did not see any existing issue for it.

This crate depends on 3 unmaintained crates:

CoreFoundation-sys: https://github.com/dcuddeback/core-foundation-sys/issues/11
IOKit-sys: https://github.com/dcuddeback/iokit-sys/issues/13
mach: https://github.com/fitzgen/mach/issues/63, [RUSTSEC-2020-0168](https://rustsec.org/advisories/RUSTSEC-2020-0168.html)

```
Crate:     mach
Version:   0.1.2
Warning:   unmaintained
Title:     mach is unmaintained
Date:      2020-07-14
ID:        RUSTSEC-2020-0168
URL:       https://rustsec.org/advisories/RUSTSEC-2020-0168
Dependency tree:
mach 0.1.2
├── IOKit-sys 0.1.5
│   └── serialport 4.2.2-alpha.0
└── CoreFoundation-sys 0.1.4
    ├── serialport 4.2.2-alpha.0
    └── IOKit-sys 0.1.5
```

sirhcel, 2023-08-23:
Thank you for reporting this issue @newAM! Do you know if there are (de-facto) successors to these crates? Especially to IOKit-sys?

newAM, 2023-08-24:
For `mach` there is [`mach2`](https://github.com/JohnTitor/mach2/tree/main).
For `IOKit-sys` and `CoreFoundation-sys` the leading alternative is [`apple-sys`](https://crates.io/crates/apple-sys).

sirhcel, 2023-09-17:
> [...] For `IOKit-sys` and `CoreFoundation-sys` the leading alternative is [`apple-sys`](https://crates.io/crates/apple-sys).

Thank you pointing out these alternatives @newAM! Having a first glance at `apple-sys`, makes it stick out from other sys crates I know by generating is code on the fly where I was expecting a crate just containing the actual bindings.

As far as I'm overseeing this for now, we could experience issues at build time from an incompatible change in the APIs exposed from Apple's SDK but no breaking API from `serialport-rs`. I'm considering the former as having no impact as the APIs in question should be stable. If things get removed from these APIs, then we would have to handle this anyway. On top, we are using IOKit and CoreFoundation only for enumerating ports which also limits potential impact.

As we are not exposing anything from IOKit and CoreFoundation we are safe with respect to semver compatibility when Apple extends APIs with newer Framework versions.

We would lose the ability to build `serialport-rs` on a macOS system which does not have XCode command line tools installed. But it looks to me that this is no practical use case.

Has anyone of you already used this crate in production code and some experience?

> I've gone ahead with using `apple-sys` here #125

Thank you @tommy-gilligan for this PR! There are some build issues as of now but nothing which looks like we couldn't iron out quickly.

> Another alternative https://github.com/servo/core-foundation-rs

I've seen this one as well but it seems to just provide bindings for Core Foundation and not for IOKit. I found no alternative for the latter at a first glance and https://github.com/dcuddeback/iokit-sys/issues/13 has gone unanswered for about half a year. Does anyone of you know of one?

newAM, 2023-09-17:
> We would lose the ability to build `serialport-rs` on a macOS system which does not have XCode command line tools installed. But it looks to me that this is no practical use case.

Adding a build-time dependency is a minor annoyance for package maintainers (e.g. nixpkgs for MacOS), but this is a pretty common requirement for building MacOS packages.  I doubt it would cause anyone problems.

sirhcel, 2023-09-19:
> This might be obvious but is there a reason why `apple-sys` doesn't release versioned crates of generated code? Is this something it could do?

It looks to me that not including the binding code is a proclaimed goal of `apple-sys`: just maintaining the binding generator tools and not the actual generated code with will change over time.

>     * Another alternative that I haven't tried yet: https://github.com/jtakakura/io-kit-rs

This crate looks like the replacement for `IOKit-sys` I was looking for. I bet I missed it when looking at crates.io due to the dash in its name.

I would prefer using the curated `core-foundation-sys` in combination with `io-kit-sys` instead of `apple-sys` and will have a look a them the next days. I will also look into the other points and come back later.

sirhcel, 2023-09-22:
Just a quick update on using `core-foundation-sys` and `io-kit-sys` a spin in a [test branch](https://github.com/sirhcel/serialport-rs/tree/test/use-maintained-cf-and-iokit) and it builds and enumerates ports.

* The constness of the parameter `plane` of `IORegistry...` functions does not match my expectation for passing const values there but could be changed easily
* The missings I/O registry could be added easily as well

I've filed two PRs for io-kit-sys. Let's see how this plays out there.

PR #136 (2023-11-21): Use maintained sys crates for CoreFoundation and IOKit
* This is an alternative to #135 which uses creates where the actual code is in the repository and not generated during build time
* Fixes #112
* This PR drive-by fixes the use of `CFDictionaryGet` and `CFGetTypeID`
    * The former could return a `NULL` pointer which [does not seem to be accepted by `CFGetTypeID`](https://github.com/opensource-apple/CF/blob/3cc41a76b1491f50813e28a4ec09954ffa359e6f/CFRuntime.c#L500) by switching to `CFDictionaryGetIfPresent`
    * Using `CFDictionaryGetIfPresent` also allow us to use the already released v [... cut]
closed 2023-11-21 by eldruin