serialport/serialport-rs #116: Type of a serialport

DavidLibault, 2023-08-03:
If I open a serial port with

```
serialport::new(port_info.port_name, 9600).open()
```

the return type is a `Result<Box<dyn SerialPort>>` so that, depending on the platform this code if built for, the exact boxed type will be different. Fine.

I want to implement a struct to store (many things, and) the opened port for later reading/writing. The structure will look like :

```
use serialport ;

struct MyNiceStruct {
    port: Box<dyn serialport::SerialPort>
}
```

But how can I initialize the `port` member of the `MyNiceStruct` structure to some value before I open a new port ?

I would do this :

```
impl MyNiceStruct {
    fn new() -> MyNiceStruct {
        MyNiceStruct {
            port: Box::new(<dyn serial port::SerialPort>::new())
        }
    }
}
```

but the SerialPort trait doesn't implement new (which makes sense). And I can not initialize the structure without initializing the port member.

I could store a `SerialPortBuilder` in the `MyNiceStruct` instead, but, it wouldn't help : I would have to open a serial port every time I want to read or write to it, which would be really inefficient.

This question is probably purely "rust related", but I just can't figure the solution out... Maybe it could be worth adding this to the Readme.md Usage chapter, because normally you don't open a serial port, read or write to it and then get out of scope.

Thanks for your kind help.

David.

mlsvrts, 2023-08-03:
Since serialport-rs uses a RAII philosophy, if you have access to a port it is also open.

I think this results in the issue you describe, so I might suggest something like: 

```rust
port: Option<Box<dyn SerialPort>>
```

When you initialize your structure, the port will be `None`; when you acquire/open a port it will be `Some(Box<dyn SerialPort>)`.

This does create some boilerplate when reading/writing to the port, since you need to be checking if the port is `Some` -- but I think this is conceptually the same as needing to check that a port is 'open'. 

The other option, of course, is simply to open/create the port when you create your structure.

DavidLibault, 2023-08-03:
Thank you so much for your quick and explicit answer. I am having such a hard time getting used to rust...