serialport/serialport-rs #119: Read  always timeout in arm-unknown-linux-musleabihf

pandakka, 2023-08-07:
I  try to use this Create in my embedded linux board programing.  The port opened  and  send the message(I reviced  by  using the “cat /dev/ttyS3”)  But when use test funtion read() its always get  timeout Errors no matter how long Timeout() set. Did I miss something?
here is my test code :
(opnenative() have the same problem.)
[image]

cargo build :

[image]

sirhcel, 2023-08-07:
Hello @pandakka, this looks strange indeed. If there is data on the rx line after the port gets opened, than I would expect it to be received.

Just the overcautious question: Is there actually data sent to the serial port you are running the snipped above on?

Could you please cross-check the reception of data with this serial port using a second adapter, crossed RX and TX lines, and [`hardware_check.rs`](https://github.com/serialport/serialport-rs/blob/22f854dbd43966abb6be216db686a06074ccdef1/examples/hardware_check.rs) from the latest main branch? For example:

```
$ cargo run --release --example hardware_check -- --loopback-port /dev/tty.usbserial-11230 /dev/tty.usbserial-11240
```

Transmitting data back and forth between these two serial ports, should report success like (at least for the test cases not using flow control):

```
Testing paired ports '/dev/tty.usbserial-11240' and '/dev/tty.usbserial-11230':
  Transmitting from /dev/tty.usbserial-11240 to /dev/tty.usbserial-11230...
     At 2000000,8,n,1,noflow...
        success
        success
     At 115200,8,n,1,noflow...
        success
        success
     At 57600,8,n,1,noflow...
        success
        success
     At 10000,8,n,1,noflow...
        success
        success
     At 9600,8,n,1,noflow...
        success
        success
     At 9600,8,n,1,softflow...
        success
        success
     At 9600,8,n,1,hardflow...
        success
        success
```

pandakka, 2023-08-08:
@sirhcel  Thank for your help !  Thank you for your advice， that I find  my embedded linux board have a little diffent from PC,which need take some special setting :sweat:      
Now it works.  Thanks again : )

sirhcel, 2023-08-08:
I'm glad to hear that! So there is nothing more to do here.

May I just ask out of curiosity what finally caused the data not being received?

closed 2023-08-08 by sirhcel

pandakka, 2023-08-09:
Maybe  is my first time touch embedded programing,  I  try a lot  and find the path is  /dev/ttymxc3 😂. 
 By the way  available_ports() works well on my pc .But on my board side file style in "/sys/class/tty/ttymxc3/device/* is different ,which may cause the function() can't find the port ,so i need  try by my hand😂.

sirhcel, 2023-08-09:
This sounds like you are running into #66 on your embedded device. On Linux without libudev, the enumeration is done using sysfs and apparently the paths from `/sys` and not the actual devices files from `/dev` are returned.

What's your environment on the target? Is udev really not available there? How are you buildilng serialport-rs and your application for the embedded target? Could you please leave a comment in #66 if you are building without udev support?