serialport/serialport-rs #124: Serial write hangs up application on some Windows machines

bastian2001, 2023-09-08:
Apart from the issue that on some Windows machines (both my laptop and my PC), `read` just blocks the program, also the `write` function blocks the program on my laptop. It works fine on my PC, but on my Laptop it also doesn't write. I tried setting the timeout to 0, still it hangs up.

I tried using the [fork by LukaOber](https://github.com/LukaOber/serialport-rs), and that fork fixes the read function, but not the write function. I have verified that the write function is not just broken in that fork, but also in this one.

By "blocking" or "hanging up" I mean, that the write function (and in this repo also the read function, as already discussed in #29 ) never returns, not even an error. The program will just stall, regardless of the set timeout.

bastian2001, 2023-09-08:
To give some context: this is my code, with different timeouts specified, all of which stalling the program:
```rust
use serialport::new;
use std::io;

fn main() {
    let mut port = new("COM8", 38400)
        .timeout(std::time::Duration::from_millis(100))
        .open()
        .expect("Failed to open port");
    let mut serial_buf: Vec<u8> = vec![0; 100];
    loop {
        println!("Reading");
        let mut s = String::new();
        let res = io::stdin().read_line(&mut s);
        match res {
            Ok(_) => {
                println!("Read {} bytes from stdin", s.len());
                println!("Content: {}", s);
                let s = s.as_bytes();
                port.write(s).expect("Write failed");
                //^^^^^^^^ stalls here
                println!("Wrote {} bytes to Serial", s.len());
            }
            Err(e) => eprintln!("{:?}", e),
        }
    }
}
```

FrankLaterza, 2023-09-18:
I'm having the same issue. I have two threads for reading and writing using a clone of the serial port. I can't write anything on the port because the read is blocking and waiting for a response. This is not an issue on my Linux machine.

bastian2001, 2023-10-01:
> because the read is blocking

That's actually a different issue than mine. This is about the write function. You should be able to use the fork by LukaOber that I linked above. It fixes the read function. In this case, it's the write function that stalls separately from the read function. I've had stalled reads on mine too, but they were resolved when I just used the fork (there's already a merge request for this fork).

FrankLaterza, 2023-10-01:
After I changed my crate to the fork by LukaOber it worked great! I had to change some code, but this solution works for me. Thanks for sharing!

Resonanz, 2024-01-22:
So.... is this fixed now ?

bastian2001, 2024-01-22:
For different reasons, I reinstalled Windows on my PC. Since then, it works. I don't think there was a bug fix related to this, so AFAIK it's not fixed

nskartoredjo, 2024-02-13:
I've encountered a similar issue, and with a bit of digging, it turns out that the write timeout is never set for Windows machines. I've created a [pull request](https://github.com/serialport/serialport-rs/pull/158) to resolve the issue

PR #158 (2024-02-16): Fix write timeout for COM ports
Resolves the issue where the COM-port write procedure is blocking, which is caused by a static timeout deceleration equal to zero. In general, you would suspect that the `set_timeout` builder method applies a timeout to both the read and write procedures, as the documentation would suggest.

Fixes #124
closed 2024-02-16 by eldruin