serialport/serialport-rs #129: Wrong informations with Com port

charleselie, 2023-10-17:
Hello,

I'am building an app on my PC in rust using Slint for the GUI.
This app is communicating through USB with a board with rust embedded code on it. 

On the embedded code I can change inforamtions such as "manufacturer", "serial number", .... 
But unfortunately, I can't get all those informations on my app, for example the manufacturer is always wrong.

Is there a fix for that ? or another lib to use ? 

charles-élie

sirhcel, 2023-11-20:
Thank you for reporting this issue! 

> On the embedded code I can change inforamtions such as "manufacturer", "serial number", .... But unfortunately, I can't get all those informations on my app, for example the manufacturer is always wrong.

Could you please provide an example of what `dmesg` or `lsusb -v` are reporting and what ` SerialPortInfo` is returned for the port in question? May be the cause is the same as in #134.

> Is there a fix for that ? or another lib to use ?

We have to look into it. There are other serial port abstractions in rust around. But I have not worked with them and cannot tell you off the cuff.

closed 2023-12-11 by eldruin

sirhcel, 2023-12-11:
Hello @charleselie, we've merged #137 which is likely to solve your issue. Could you please give the current master a try? You can use this revision by replacing your dependency with the [unreleased changes](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-dependencies-from-git-repositories) in `Cargo.toml`:
```
serialport = { git = "https://github.com/serialport/serialport-rs.git", rev = "e10d3c9" }
```

Feel free to reopen this issue if your problem still exists with.