serialport/serialport-rs #138: Add supported architecture

deid84, 2023-11-23:
Hello, i'm succesfully using from few weeks this crate within my STM32MP1 that has an armv7 onboard.

Do you think could be useful to add armv7-unknown-linux-gnueabihf to the supported architectures? The only difference i saw is that i need to configure my cargo toml in this way:

serialport = {version = "4.2.2", default-features = false}

in order to be able to cross-compile without having error on libudev library, that's a minor issue for me 'cause i don't need to list serial ports on my software.

Maybe you have some ideas about how to cross-compile succesfully with default features enabled

thank you!

sirhcel, 2023-12-10:
Hello @deid84, I'm glad to hear that you are successfully using serialport-rs.

> Do you think could be useful to add armv7-unknown-linux-gnueabihf to the supported architectures?

We could add this target but this would not change anything for your use case. I bet we have enabled using udev on Linux by default because this is a sensible default for people using serialport-rs on a desktop system as they usually do not have the background with such such details as people working with embedded systems do.

> The only difference i saw is that i need to configure my cargo toml in this way:
>
> serialport = {version = "4.2.2", default-features = false}
>
> in order to be able to cross-compile without having [...]

I have to admit that I have neither cross compiled with libudev manually. Because either Yocto has taken care of the build environment for me or I got annoyed before reaching the finish line. Building for libudev by default is even a sensible default for cross compiling with Yocto. What's wrong for you with disabling the default features?

On the other hand, cross compiling without libudev selects a Sysfs backend for enumerating serial ports. Did you give it a try on your target? For example with our `list_ports` example? I'm asking because a quick check on an i.MX system some days ago looked at a first glance like our heuristic with Sysfs is failing there and I got no ports enumerated there.

Would some additional words about it in `README.md` have helped you with switching to a build without default features earlier? If yes, could you make a proposal/file a PR?

> Maybe you have some ideas about how to cross-compile succesfully with default features enabled

How does your build setup looks like? And what error are you getting there?

Best regards!

deid84, 2023-12-20:
Hello @sirhcel , sorry for my late reply.

I have to admit that once compiled disabling "default-features" everything worked fine for me. 
My project involves a customized layer on a STM32MP1 so I have only 2 RS485 ports in my device tree and I never have to choose between them because they talk with to instruments that will remain in the same position forever.

My question is just in case someone needs a port enumeration also in a armv7 cross-compiled solution.

I think that your idea about giving more information on readme.md would be great, but unfortunately I don't know exactly how to help you to write some additional info. Please let me know if it could be helpful for you if I try to better describe my build setup or something else ok?

Best Regards!

chuyao, 2024-06-24:
tks, this solved my problem