serialport/serialport-rs #146: windows: UsbPortInfo: Provide the same info about Manufacturer and Product name as POSIX

jcsxdev, 2023-12-27:
1. [Issue description](#_issue_description)
2. [Improvement request](#_improvement_request)
3. [Others](#_others)

## 1. Issue description <a name ="_issue_description"></a>

On POSIX systems, the [UsbPortInfo](https://docs.rs/serialport/latest/serialport/struct.UsbPortInfo.html#) returns custom Manufacturer and Product name linked to iVendor and iProduct attributes from Device Descriptor. In contrast, in Windows:
-  the Manufacturer name is obtained from the [DEVPKEY_Device_Manufacturer](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-manufacturer) property key; and 
- the Product name is obtained from the [DEVPKEY_Device_FriendlyName](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-friendlyname) property key.

This is causing the following issue:
- The Manufacturer and Product name information obtained from Windows is completely different from POSIX. If this information is used for device detection, the code will not be portable.

## 2. Improvement request <a name ="_improvement_request"></a>

In Windows, the information must be obtained via [DeviceIoControl](https://learn.microsoft.com/en-us/windows/win32/api/ioapiset/nf-ioapiset-deviceiocontrol). In a similar way to the **lswinusb** –  a Rust library for listing USB Descriptors on Windows:
- https://github.com/noreyatech/lswinusb

Can this improvement be included in the project?

## 3. Others <a name ="_others"></a>

Additional information:
1. The [USBViewer](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview) program also uses [DeviceIoControl](https://learn.microsoft.com/en-us/windows/win32/api/ioapiset/nf-ioapiset-deviceiocontrol):
    - https://github.com/microsoft/Windows-driver-samples/blob/main/usb/usbview/enum.c
2. Python module [pySerial](https://github.com/pyserial/pyserial) also faces the same problem:
    - https://github.com/pyserial/pyserial/pull/338
    - https://github.com/pyserial/pyserial/pull/637
    - https://github.com/pyserial/pyserial/pull/725

sirhcel, 2024-01-12:
You are absolutely right. I would expect to get the same information on systems where the port enumeration returns anything at all.

Than you very much for pointing this out and your comprehensive research! My schedule is tight these days. But it looks like you are already working on implementing this and this looks good skimming through it. Awesome! :) Would you mind filing a PR when this works for you?

jcsxdev, 2024-01-15:
> Than you very much for pointing this out and your comprehensive research! My schedule is tight these days. But it looks like you are already working on implementing this and this looks good skimming through it. Awesome! :) 

Thank you for the feedback.

> Would you mind filing a PR when this works for you?

No problem. I will file a PR when I complete some tests.

Example:
- The tests I had performed worked under the following condition:

[image]

- But they failed under the following condition:

[image]

I have made new changes to the [iocontrol feature](https://github.com/jcsxdev/serialport-rs/tree/feature/iocontrol) to fix this issue, but they haven't been tested yet.