serialport/serialport-rs #148: use read_to_end always return TimedOut err

lethargy123, 2024-01-08:
I use serialport-rs on windows11, in use read_to_end(SerialPort tarit) always return TimedOut err
err:called `Result::unwrap()` on an `Err` value: Custom { kind: TimedOut, error: "Operation timed out" }

sirhcel, 2024-01-12:
Thank you for bringing this up! [`Read::read_to_end`](https://doc.rust-lang.org/std/io/trait.Read.html#method.read_to_end) seems like a tricky task for a serial interface because I don't see what the natural end-of-file (EOF) would be in this case. 

I bet your are encountering a phenomenon from the default implementation of `Read::read_to_end`. This might not be intuitive but it seems to do its job according to its spec. How should serialport's `read_to_end` work in your opinion? How would you define end-of-file for a serial stream?

RossSmyth, 2024-03-27:
Yeah I reached for this API as well when I first started doing serialport operations in Rust. Mainly because I was biased by pySerial & other libraries. The way I initially expected it to work was the following:

1. Set the timeout to some value
2. Call `read_to_end`
3. When the read timeout occurs, that would count as an "end"

This is similar to how the pySerial `port.read()` API works. That way reading a whole "packet" just works. Because for all of my use cases the host (the computer running Python/Rust code) drives the interactions. The device never sends an unsolicited serial message. So the loop of `Host -> Device |> Wait for reply |> Device -> Host` pretty much always works.

Recreating something like that isn't too hard.

```rust
fn read_until_timeout<const BUFF: usize>(
    port: &mut dyn SerialPort,
    length_limit: Option<NonZeroUsize>,
) -> std::io::Result<Box<[u8]>> {
    let mut output = Vec::with_capacity(BUFF);
    let mut buf = [0_u8; BUFF];

    loop {
        let res = port.read(&mut buf[..]);

        match res {
            // Returning zero means EOF. I'm not sure if a serial device can receive EOF
            Ok(0) => break Ok(output.into()),

            // Put some data in to the buffer
            Ok(fill) => {
                output.extend_from_slice(&buf[..fill]);

                if let Some(max_size) = length_limit {
                    if output.len() == max_size.into() {
                        output.truncate(max_size.into());
                        break Ok(output.into());
                    }
                }
            }
            // Timeout is a success case, so return the data.
            Err(err) if std::io::ErrorKind::TimedOut == err.kind() => break Ok(output.into()),
            Err(err) => break Err(err),
        }
    }
}
```