serialport/serialport-rs #149: Stuck when writing data to the serial port if paired port is not open

Mil0R, 2024-01-10:
I used com0com to simulate a pair of virtual serial ports com3 and com4, then I run the code below to write data to com3

1. If the com4 is open, the code runs fine
2. If the com4 is closed, the code stuck

```
use std::time::Duration;

fn main() {
    println!("Hello, world!");
    let mut port = serialport::new("com3", 9600)
        .timeout(Duration::from_millis(10))
        .open()
        .expect("Failed to open port");
    let output01 = "This is a test. This is only a test. 01".as_bytes();
    port.write_all(output01).expect("Write failed! 01");
    let output02 = "This is a test. This is only a test. 02".as_bytes();
    port.write_all(output02).expect("Write failed! 02");
    let output03 = "This is a test. This is only a test. 03".as_bytes();
    port.write_all(output03).expect("Write failed! 03");
}
```

sirhcel, 2024-01-13:
Does writing work in the scenario where the receiving port is closed when using another serial framework like pySerial? Could it be the case that com0com behaves differently to a physical null-modem cable without hardware handshaking in the sense that it blocks writing data where with a real cable data would still transmitted by the sender?

Mil0R, 2024-01-14:
Yeah, that's really how com0com's handles it when there's no hardware connected, and when I connect a real piece of hardware the problem is gone.

closed 2024-01-14 by Mil0R