serialport/serialport-rs #153: USB ports detected as unknown in docker

dignifiedquire, 2024-01-16:
When running inside of docker (base image: `rust:1.75-buster`) and trying to list the detected ports, I am getting this 

```
Found 3 ports:
  /dev/ttyUSB0
    Type: Unknown
  /dev/ttyUSB1
    Type: Unknown
  /dev/ttyUSB2
    Type: Unknown
```

Things work fine when running on the same machine outside of docker. 

Any ideas how to debug/fix this?

soiamsoNG, 2024-03-28:
[udevadm-does-not-show-all-attributes-inside-a-docker-container](https://stackoverflow.com/questions/41753218/udevadm-does-not-show-all-attributes-inside-a-docker-container)

legas7, 2025-03-03:
Isn't this a case where you need to pass device into docker via `--device=/dev/ttyUSB0`? (https://stackoverflow.com/a/24231872)