serialport/serialport-rs #157: Unplug serial device cause system shutdown on windows

ArthurZhou, 2024-02-04:
Once the connection is built, power off the serial device will cause the system to shutdown.

My serial device is an NodeMCU ESP32S. It communicate with my computer through Bluetooth.

Windows Bluetooth config:
[image]

Windows version:
```
Edition	Windows 10 Enterprise LTSC
Version	21H2
Installed on	‎8/‎15/‎2023
OS build	19044.3693
Experience	Windows Feature Experience Pack 1000.19053.1000.0
```

My rust code: 
```rust
// #![windows_subsystem = "windows"]

use std::{env, fs};
use std::io::Read;
use std::path::{Path, PathBuf};
use std::time::Duration;

fn main() {
    let path = &env::current_dir().unwrap();
    let port_name = "COM6"

    let baud_rate = 115200;

    loop {
        start(path, port_name, baud_rate);
    }
}

fn start(path: &PathBuf, port_name: &str, baud_rate: u32) { // this is the receiver function
    let port = serialport::new(port_name, baud_rate)
        .timeout(Duration::from_secs(5))
        .open();

    match port {
        Ok(mut port) => {
            let mut serial_buf: Vec<u8> = vec![0; 1000];
            println!("Receiving data on {} at {} baud:", &port_name, &baud_rate);
            loop {
                match port.read(serial_buf.as_mut_slice()) {
                    Ok(t) => {
                        let s: String = String::from_utf8_lossy(&serial_buf[..t]).to_string();
                        // execute some command, ignoring the "ALIVE" message
                    },
                    Err(_) => { // the device will send an "ALIVE" message every 2 secs. if not, the device might be disconnected. 
                                    // and we shall restart the receiver function and wait for next connection
                        println!("COM port disconnected");
                        break;
                    },
                }
            }
        }
        Err(e) => {
            println!("Failed to open \"{}\". Error: {}", port_name, e);
        }
    }
}

```

sirhcel, 2024-02-05:
Hello @ArthurZhou, does this mean your application exists or that your operation system shuts down and turns the lights off?

I would expect that the application quits due to the serial port getting closed by the operating system when your Bluetooth device gets powered off. But not the latter. Are you getting the message `COM port disconnected` when powering off the Bluetooth device?

TiemenSch, 2024-03-27:
I was testing the disconnection behavior on my system, too (Surface Pro 8) and it gave me a blue screen repeatedly when disconnecting a USB device. And with disconnection I meant pulling the cable to see what happens.

ArthurZhou, 2024-03-28:
> I was testing the disconnection behavior on my system, too (Surface Pro 8) and it gave me a blue screen repeatedly when disconnecting a USB device. And with disconnection I meant pulling the cable to see what happens.  

Maybe a windows bug, as I cannot reproduce it on Ubuntu desktop and arch linux.

But this doesn't happened when I use Python to do this on windows

Weird🤔

TiemenSch, 2024-03-29:
Yeah, never had any troubles with the same executable on my Ubuntu desktop either. 

My Windows BSOD sequence:
1. good connection and operation
  - I re-open the connection periodically to receive some data and then close again.
2. pull plug
3. plug back in
4. BSOD 

Where at step 2 and 3 I often receive some disconnect error or a subsequent "no rights" kind of error for the replugged device. I wouldn't mind an error or even a panic, but a full BSOD is kind of heavy hitting.

Could it be related to the `x86_64-pc-windows-gnu` target? I don't know the ins and outs of the target-specific dependencies/implementations.

TiemenSch, 2024-07-29:
Is there any progress on this? What can I do to help pinpoint the problem?

ArthurZhou, 2024-08-03:
> > Is there any progress on this? What can I do to help pinpoint the problem?
> 
> 
> 
> Not from my side due to way lower hanging fruits.
> 
> 
> 
> A log or screenshot of the BSOD and an minimal working example could (like in `exapmples/` in a fork of this project) could be a good start. Additionally the Python counterpart of this example. And finally some insights in how to trace "syscalls" on Windows for comparing setup and actions in both cases.

currently i am unable to reproduce this bug after i updated my windows.

i've tried to reproduce this bug on Windows 10 LTSC 2021, with latest system bug fixes installed (2024/08/03)

i will try to collect and post the bsod info and other detailed info here if this happens again

(sorry for my poor english

closed 2026-11-15 by ArthurZhou