serialport/serialport-rs #161: Fast receive missing beginning of packet

usbalbin, 2024-02-23:
Hi

I am building an embedded device which needs to log a lot of data quite fast to a PC. Right now I have hardcoded the device to send the same 52bytes of data every 2ms, this works fine as seen on logic analyzer.

Now on the PC end, I am using this library to try and receive that data. Since the data is [COBS](https://en.wikipedia.org/wiki/Consistent_Overhead_Byte_Stuffing) i thought I'd use `BufReader::split(SENTINEL_BYTE)`. However for some reason every about 9-10th received packet is missing its first x bytes, where x varies.

```rust
fn main() {
    use std::io::{BufReader, BufRead};

    // This is what the device prints(including the 0 byte) which BufReader::split removes as expected
    const EXPECTED_DATA: [u8; 52 - 1] = [
        1, 10, 5, 29, 1, 25, 2, 248, 85, 51, 9, 180, 16, 176, 9, 182, 2, 51, 3, 6, 180,
        36, 248, 172, 1, 128, 128, 192, 247, 13, 180, 36, 224, 93, 52, 10, 152, 17,
        148, 10, 192, 2, 52, 40, 134, 3, 194, 222, 152, 241, 167,// 0 <- End byte
    ];
    let baud = 460800;
    let port = serialport::new("/dev/ttyUSB0", baud).open_native().unwrap();
    let mut reader = BufReader::with_capacity(u16::MAX as _, port);
    let mut successes_in_a_row: u64 = 0;
    println!("Prints successed received bytes in a row on failure:");
    for data in reader.split(0) {
        let Ok(data) = data else {
            continue
        };
        if data == EXPECTED_DATA {
            successes_in_a_row += 1;
        } else {
            println!("{successes_in_a_row}, {}B", data.len());
            successes_in_a_row = 0;
        }
    }
}
```

output from cargo r --release
```
Prints successed received bytes in a row on failure:
0, 65B
9, 33B
8, 5B
9, 29B
8, 1B
9, 25B
9, 49B
8, 21B
9, 45B
8, 17B
9, 41B
8, 13B
9, 37B
8, 9B
9, 33B
8, 5B
9, 29B
8, 1B
9, 25B
9, 49B
8, 21B
9, 45B
8, 17B
9, 41B
8, 13B
9, 37B
8, 9B
9, 33B
8, 5B
9, 29B
8, 1B
9, 25B
9, 49B
8, 21B
9, 45B
8, 17B
9, 41B
8, 13B
9, 37B
8, 9B
9, 33B
8, 5B
9, 29B
...
```

Any ideas as to what could be the cause? Some sort of buffer overrun? I have yet to see any wrong bytes in the stream, it is just that data sometimes is missing its first set of bytes

torkleyy, 2024-03-04:
I've been experiencing the same issue, we couldn't figure out what's causing it. However, some tools we use don't lose any of the UART data...

DanielJoyce, 2024-04-10:
Instead of swallowing the error, can you print it out? It might be something obvious.

```
  let Ok(data) = data else {
            continue
        };
```

This continues the next iter of the loop, and doesn't print anything out nor reset the success counter.

henm, 2024-05-15:
I'm experiencing a similar issue. I've noticed that in my case version 4.0.1 works, so I'm using that one for the moment.

sirhcel, 2025-09-16:
Alright, looks like this is nothing which automatically went away with 4.0.1. I would like to reproduce this issue:

* What operating system (and version) are/were you seeing this issue on @rsmyth-ecmi,  @henm, @torkleyy, @usbalbin?
* Are you seeing this issue just with USB serial devices or other interfaces as well?
* How could a scenario for reproducing this issue look like?