serialport/serialport-rs #179: Timeout reading SerialUSB from Arduino DUE or GIGA

dbalsom, 2024-04-28:
The Arduino MEGA has a UART implementation (probably something like [FTDI](https://ftdichip.com/)) 
The Arduino DUE and GIGA boards have native USB serial interfaces.

serialport-rs can enumerate, open, read and write the port from a MEGA. 

serialport-rs can enumerate, open the serial port on a DUE or GIGA, but calling read() results in a timeout, even if the Arduino is spinning in a loop printing "hello world" to the SerialUSB interface; so data should always be available.

The units work fine when interfacing via pyserial in a python program. 

Baud rate should be irrelevant for the native USB interface but I have tried several values, as well as playing with stop bits and parity. 

I'm not sure what else might be helpful in troubleshooting but I'd be happy to provide what I can. If someone needs a GIGA sent to them for troubleshooting I'd be willing to arrange it.

dbalsom, 2024-04-28:
Also fixed by https://github.com/serialport/serialport-rs/pull/94 .

sirhcel, 2024-04-29:
Thank you for bringing up this issue and especially the setup for reproducing it! Thank you very much for offering to ship a prepared Arduino device as this will reduce the initial effort to start looking into this dramatically. I would like to look into it and would take your offer.

Could you please prepare an Arduino device with the firmware you described above? Could you please also prepare a repository with the host application using serialport-rs and the sketch on the device? I've send you details for shipment via mail.

dbalsom, 2024-04-29:
Sorry if my reply wasn't clear - this issue is fixed by the linked PR