serialport/serialport-rs #182: Opening a COM port on Windows fails

miguelraz, 2024-05-06:
If I run
```rust
fn main() {
    let ports = serialport::available_ports().expect("No ports found!");
    for p in ports {
        println!("{}", p.port_name);
    }
    // COM8 was printed for me
    serialport::new("COM8", 9600).open().expect("we done goof'd");
}
```
I get
```rust
❯ cargo run
   Compiling serialbug v0.1.0 (C:\Users\mrg\work\serialbug)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 3.01s
     Running `target\debug\serialbug.exe`
COM8
thread 'main' panicked at src\main.rs:7:42:
we done goof'd: Error { kind: NoDevice, description: "The system cannot find the file specified." }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: process didn't exit successfully: `target\debug\serialbug.exe` (exit code: 101)
```

mlsvrts, 2024-05-06:
This shouldn't be required, but you could try the 'escaped' syntax: `\\.\COM10`

Other thoughts:

- Does it work if you try from an admin command prompt?
- I assume you can open the port in a program like PuTTy
- Can you providenthe error returned?

miguelraz, 2024-05-06:
Hey! 
* I tried using `r"\\.\COM8" and it didn't work either.
* Yes, running from an admin command prompt also fails.
* I haven't tried PuTTy yet
* yes, it shows up exactly like the screenshots here https://github.com/zephyrproject-rtos/zephyr/issues/17266

miguelraz, 2024-05-06:
Connecting via PuTTy also failed:
[image]

after 
[image]

mlsvrts, 2024-05-06:
> Hey! 
> * I tried using `r"\\.\COM8" and it didn't work either.
> * Yes, running from an admin command prompt also fails.
> * I haven't tried PuTTy yet
> * yes, it shows up exactly like the screenshots here https://github.com/zephyrproject-rtos/zephyr/issues/17266 and with the same error messages

Unfortunately this is most likely an issue with your device drivers/USB descriptors, and isn't something that can be fixed by this library.

These issues can be tricky to debug, but I recommend taking a look at the device descriptor using [USBTreeView](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview) as a place to start.

miguelraz, 2024-05-06:
Thanks, will close this issue then.

closed 2024-05-06 by miguelraz