serialport/serialport-rs #187: Cannot list virtual serial ports on Windows 10

golanguage, 2024-05-12:
[com0com](http://com0com.sourceforge.net/)  [mirror](https://github.com/paulakg4/com0com) Null-modem emulator is a long history kernel-mode virtual serial port driver for Windows.  

But these port cannot be detected by this lib;
```
serialport-rs on  main is 📦 v4.3.1-alpha.0 via 🦀 v1.78.0
❯ cargo run --example list_ports
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 1.56s
     Running `target\debug\examples\list_ports.exe`
No ports found.
```

I can find its port using WIN10 system serial tool:
```
❯ [System.IO.Ports.SerialPort]::getportnames()
CNCA0
CNCB0
```

Or anothere Rust serial port lib [serial2](https://github.com/de-vri-es/serial2-rs).
```
serial2-rs on  main is 📦 v0.2.24 via 🦀 v1.78.0
❯ cargo run --example list-ports
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 1.09s
     Running `target\debug\examples\list-ports.exe`
Found 2 ports
CNCA0
CNCB0
```

sirhcel, 2024-05-12:
Thank you for bringing up this issue! You are mentioning _serial2_ which uses different enumeration strategy on Windows. There is #84 in our PR backlog since a while, which switches to the same strategy and it looks like this could resolve your issue as well.

I would like to integrate it with with a slight change as prototyped in https://github.com/sirhcel/serialport-rs/tree/missing-modem-ports (as of [0720d5a](https://github.com/sirhcel/serialport-rs/commits/0720d5a64109d43b2d5fa5fb170e3ed052dbf9b1/)). Could you give this branch a spin? Will it make the null-modem ports you are looking for getting listed? And what about regular serial ports: do they show up as before and as you would expect them to?

golanguage, 2024-05-12:
seem it's a great job.
In fact, there is not a real serial port on the PC, all 4 ports are virtual.

```
serialport-rs on  missing-modem-ports is 📦 v4.3.1-alpha.0 via 🦀 v1.78.0
❯ cargo run --example list_ports
   Compiling serialport v4.3.1-alpha.0 (C:\proj\rust\serialport-rs)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 6.10s
     Running `target\debug\examples\list_ports.exe`
Found 4 ports:
  COM3
    Type: Unknown
  COM4
    Type: Unknown
  CNCA0
    Type: Unknown
  CNCB0
    Type: Unknown
```

> Thank you for bringing up this issue! You are mentioning _serial2_ which uses different enumeration strategy on Windows. There is #84 in our PR backlog since a while, which switches to the same [...]

sirhcel, 2024-05-12:
Solved with integrating #84.

closed 2024-05-12 by sirhcel