serialport/serialport-rs #206: How to close the open port?

krishnaTORQUE, 2024-07-28:
How to close / disconnect the open port?

sirhcel, 2024-07-28:
It is closed when it gets [dropped](https://docs.rs/serialport/latest/serialport/struct.TTYPort.html). For example by leaving the scope or explicitly dropping it with [`drop(your_port_here)`](https://doc.rust-lang.org/std/mem/fn.drop.html) if you want to close it earlier.

krishnaTORQUE, 2024-07-29:
Great, thanks

closed 2024-07-29 by krishnaTORQUE

sirhcel, 2024-07-29:
You are welcome! We've had this question before and it looks like we could state this more clearly:

* It is stated under _Usage_ in [`README.md`](https://github.com/serialport/serialport-rs/blob/b0d046b3e1b52146987756d4efab3d9e69ddd0b8/README.md#usage) but in contrast to other aspects without an example
* The generated [overview docs](https://github.com/serialport/serialport-rs/blob/b0d046b3e1b52146987756d4efab3d9e69ddd0b8/src/lib.rs#L7) just briefly mention opening a serial port but not closing it

Which docs (if any at all) did you consult in the first place? May be we should reopen this issue for cleaning up documentation.