serialport/serialport-rs #211: Please add build instructions

Resonanz, 2024-08-23:
Please add the following noob build instructions into the Examples section of README.md:

How to run example code, for example "list_ports.rs": 

```
cargo run --example list_ports
```

RossSmyth, 2024-08-23:
I don't think there's any reason to. That is a Cargo concern. One will learn how just by learning Rust or typing `cargo --help`. 

https://doc.rust-lang.org/cargo/reference/cargo-targets.html

sirhcel, 2024-08-31:
You are right @RossSmyth, this is about Rust basics.

On the other hand are we already providing an example of how to build the crate without default features in the README's [Overview](https://github.com/serialport/serialport-rs/blob/dc8276a1c1fad77914a55ac146ddf22121abf001/README.md#overview) section. So I'm not against adding another example for `list_ports` there for a quick start with quick success.

Please feel free to open a PR @Resonanz.