serialport/serialport-rs #226: Add posibility to enable rs485 functionality for port under Linux

slyshykO, 2024-10-31:
There is [rs485](https://crates.io/crates/rs485) that provides this functionality. 
But it has not been updated for 7 years.
And it is convenient to have such functionality right in place.

sirhcel, 2024-11-24:
Thank you for this suggestion! Yes, it would be convenient to support this functionality directly from the serialport crate. 

My gut instinct tells me that verifying this functionality will become the hardest part. Do you know of some commodity hardware which we could use for testing? Especially the more exotic stuff like activating a termination resistor?

olback, 2025-02-11:
> Thank you for this suggestion! Yes, it would be convenient to support this functionality directly from the serialport crate.
>
> My gut instinct tells me that verifying this functionality will become [...]

We're using Moxa devices in a lot of our products, specifically, the UPort 1450I, and they work great. They have DIP-switches internally to control the termination resistors. Everything in this reply below will be about said device, for context.

[image]
([Moxa UPort Manual](https://www.moxa.com/getmedia/73259f0a-12b4-4822-ac5f-319bf000c2bb/moxa-uport-1200-1400-1600-series-manual-v8.1.pdf))

Here's what I'm currently working with:
```
$ lsusb
[..]
Bus 001 Device 005: ID 110a:1450 Moxa Technologies Co., Ltd. UPort 1450 4-Port RS-232/422/485
[..]
```

I first tried to use the rs485-crate like OP mentioned, but the `mxuport`-driver does not support that IOCTL call. Instead, you change the `port` value in the `serial_port` struct.

```rust
// See `serial_struct` in the Linux kernel source: https://github.com/torvalds/linux/blob/master/include/uapi/linux/serial.h
#[repr(C)]
#[derive(Debug, Clone)]
struct LinuxSerialStruct { .. }

let mut sp = serialport::new(port, baud)
    .open_native()
    .expect("Failed to open serial port");

// Read the serial struct
let mut ss_old = LinuxSerialStruct::default();
if unsafe { libc::ioctl(sp.as_raw_fd(), libc::TIOCGSERIAL, &mut ss_old) } < 0 {
    eprintln!("Failed to GET serial struct");
    return;
}
println!("Old serial struct: {:#?}", ss_old);

// Set the serial struct with modified port
let mut ss_new = ss_old.clone();
// RS232 = 0, RS485 2W = 1, RS422 = 2, RS485 4W = 3 as per the MOXA UPort Manual
ss_new.port = 1; // RS485 2W

if unsafe { libc::ioctl(sp.as_raw_fd(), libc::TIOCSSERIAL, &mut ss_new) } < 0 {
    eprintln!("Failed to SET serial struct");
    return;
}

let dyn_sp = &mut sp as &mut dyn serialport::SerialPort;

// Use dyn_sp like normal, but make sure to not TX/RX at the same time!
```

All this to say that all serial port devices are unique in handling configuration. Moxa uses TIOCSSERIAL and the rs485-crate uses TIOCSRS485.