serialport/serialport-rs #231: Will rust 1.59.0 be a barrier to using the newer create?

sweet-de-way, 2024-12-02:
Over time, many third-party libraries are updated to take advantage of new features and improvements in Rust. If your project depends on these libraries and they no longer support Rust 1.59.0, using an older version of the compiler may become a problem.

For example, windows-sys 0.59.0 requires libc 0.2.164 to build on some platforms.

#228

sirhcel, 2024-12-08:
Yes, using MSRV 1.59.0 is already a barrier and this is getting more of a challenge as rust releases progress. The reason we are using this "prehistoric" release as our MSRV is the latest long-term support release of Yocto (for building embedded Linux distributions) provided this Rust release.

A new long-term support release of Yocto is out and I wanted to tackle this long overdue bump of the MSRV together with a bigger refactoring of the interface (like #34, #94 ...). But it looks like this is too much asked and not gonna happen soon.

I consider bumping our MSRV as a semver breaking change. The seralport crate stumbled more than one time over dependencies bumping their MSRV with a minor release and I'd like to spare users this not so pleasant experience. That's why I'm starting to get inclined to do split this up in two major releases:

* Release 5.0 for bumping the MSRV to 1.75.0 (which is the one supported by the current long-term support release of Yocto)
* Releaso 6.0 for transitioning to a platform-specific representation of `SerialPort`

So a switch to windows-sys could land with each of them.

unlimitedsola, 2024-12-12:
> * Release 5.0 for bumping the MSRV to 1.75.0 (which is the one supported by the current long-term support release of Yocto)

Hi, I'm also seeing a lot of PRs are currently blocked by this MSRV requirement, so I wonder when could this happen? Should we start branching out with a newer MSRV so that we can start getting these PRs merged?

sirhcel, 2024-12-12:
> Hi, I'm also seeing a lot of PRs are currently blocked by this MSRV requirement, so I wonder when could this happen? Should we start branching out with a newer MSRV so that we can start getting PRs merged and have a foundation to work on?

Which PRs are you referring to here? For sure, the list of PRs and also tagging them. As of now both, semver breaking and MSRV breaking changes are tagged as _5.0.0_. We could and likely should split this up between semver and MSRV breaking changes.

unlimitedsola, 2024-12-13:
> Which PRs are you referring to here?

I'm referring to PRs that are tagged with _5.0.0_. There are currently 8 open PRs that are tagged with _5.0.0_, and the work for migrating to `window-sys` will likely be one too, due to the amount of cleanup we need after we bumped MSRV if we stay 1.59.0.

> We could and likely should split this up between semver and MSRV breaking changes.

The question is how should we proceed? Should we start merging MSRV breaking changes into `main`? **or** should we at some point bump the MSRV to 1.75.0 (as you've said the Yocto project is the main deciding factor anyways) on `main` first so that pending PRs can have a working foundation without also adjusting the MSRV themself and causing conflicts with each other? and when could this happen? Are there issues that we have to resolve before bumping MSRV?

fpagliughi, 2025-01-24:
I'll pile on here, too. Using Yocto, we've learned to put _Cargo.lock_ files into our repo for all of our internal Rust crates to ensure that MSRV updates to our dependencies don't break our build. 

I assume that everyone using Rust on Yocto has learned this lesson?

But that has also freed me to bump the MSRV on my own open-source hw/embedded Rust crates without worrying about breaking my own builds and everyone else's. I love that cargo added a Minimum RSV, but I wish they would also add a Maximum RSV to resolve the latest version of dependencies that don't break the build. 

I think the latest Yocto Rust version is a good target for MSRV in a crate like this. MSRV 1.75 would be _really_ helpful right now.

paolobarbolini, 2025-04-30:
As time moves on more and more dependencies bump MSRV. Even `tokio` has been requiring >1.59 for 1.5 years. Is there any chance that with the new [MSRV aware resolver](https://blog.rust-lang.org/2025/01/09/Rust-1.84.0/#cargo-considers-rust-versions-for-dependency-version-selection) we could relax the MSRV requirement?

As @fpagliughi said in Rust it's very important to lock your dependencies by committing `Cargo.lock` to your repositories. This not only helps with this but it's good in general for reproducibility and controlling _when_ things are updated.

I also find the Rust versions in Yocto to be highly outdated and unusable in many cases. It's comparable to trying to get the current Rust ecosystem to build on Debian Bookworm (Rust v1.63), except that for them at least a good part of the community still tries to care for.
Anyways, maybe this crate could be like tokio: multiple minor versions are supported. Only important fixes are backported (I'd also make it _on request_) and at least this way development can move forward and not be blocked by a 3 years old OS release (although I get that most of the community is still stuck on it 🙄).

sirhcel, 2026-03-23:
We unpinned dependencies with #328 and cut the release [4.9.0](https://crates.io/crates/serialport/4.9.0). While still supporting Rust 1.59.0 you are no longer limited by pinned dependencies. Finally!