serialport/serialport-rs #237: Write doesn't get sent

Skyfrei, 2024-12-30:
```rs
fn write_serial(port: &mut Box<dyn SerialPort>) {
    println!("\nWrite the serial message!\n");
    let mut buffer = String::from("\rVERSIONA\r\n");
    //let _ = io::stdin().read_line(&mut buffer);
    let buffer_trimmed = buffer.trim();

    if buffer_trimmed == ""{
        return;
    }
    else if buffer_trimmed == "exit"{
        close_port(port);
        kill_program(0);
    }
    let write_buffer = buffer_trimmed.as_bytes();
    println!("{:?}", write_buffer);
    let res = port.write(write_buffer);
    check_for_errors(res);
}
```

This is the program i use to write to the serial port. I can read from the port fine, and i dont get any errors when i write to the device. However, the write isn't going through? Or something else is happening.  

This is how i am configuring the port.
```
let serial_settings = SerialCommunicationSettings { comm_speed: baudrate };
    let mut port = serialport::new(port_name, serial_settings.comm_speed)
        .data_bits(DataBits::Eight)
        .stop_bits(StopBits::One)
        .flow_control(serialport::FlowControl::None)
        .timeout(Duration::from_millis(3000))
        .open()
        .expect("Failed to open port");

    let _ = port.write_data_terminal_ready(true);
    let _ = port.write_request_to_send(true);

```

Device does respond as this works with another library.

sirhcel, 2024-12-30:
At a first glance, the data to be written should appear on the wire without any issues. May be `write_serial` just misses flushing the data and its' still stuck in library/OS buffers? Could you give explicitly flushing the data a try?

The port configuration looks fine and in my quick test [`examples/issue_237_tx_test.rs`](https://github.com/sirhcel/serialport-rs/commit/42e41b572cae1884db74a09b63ab5167ae7f2c59), the data appears just fine on the wire.

Skyfrei, 2024-12-31:
I will have to try again with flushing. I am using a unicore UM982 to test (which is the error above).

But testing with an arduino R4 minima it works fine without flushing. I will give it another try in 2 days and update this.

Skyfrei, 2025-01-02:
Nope, still nothing shows up. The UM982 is working because i am getting messages to read. However, writing doesn't seem to do anything to it.

## Read msgs
```md
0,18,1;INSUFFICIENT_OBS,NONE,0.00000000000,0.00000000000,0.0000,0.0000,WGS84,0.0000,0.0000,0.0000,"",0.000,0.000,0,0,0,0,0,00,00,00,INSUFFICIENT_OBS,NONE,0.000,0.000,0.0000,0.000000,0.0000,0.0000,0.0000*209e4607
$GNGGA,,,,,,0,,,,,,,,*78
$GPGSA,,1,,,,,,,,,,,,,,,,*73
$GNRMC,,V,,,,,,,,,,N,V*37
$GPGST,,,,,,,,*57
$GPGSV,1,1,00,,,,,*55
#PVTSLNA,96,GPS,UNKNOWN,1,460000,0,0,18,2;NONE,0.0000,0.00000000000,0.00000000000,0.0000,0.0000,0.0000,0.000,NONE,0.0000,0.00000000000,0.00000000000,0.0000,0=============================================================================================================
```

but upon writing a command i should get something like command OK, and a response from that command.

Skyfrei, 2025-01-02:
Fixed it, i think the issue wasn't the write  itself but the formatting.
```rs
let write_buffer = format!("{}\r\n", buffer_trimmed).into_bytes();
    match port.write_all(&write_buffer) {
        Ok(_) => {
            match port.flush() {
                Ok(_) => println!("Message sent successfully"),
                Err(e) => println!("Failed to flush port: {}", e)
            }
        },
        Err(e) => println!("Failed to write to port: {}", e)
    }
```

Since as_bytes formats with a terminator at the end but no new line in the beginning im assuming the microcontroller o the UM982 wasn't seperating the read from the write and it was being read as 1 line.

closed 2025-01-02 by Skyfrei

sirhcel, 2025-01-02:
> Fixed it, i think the issue wasn't the write itself but the formatting.

Glad to hear that. Happy new year and thank you for keeping us posted here!

Skyfrei, 2025-01-02:
> > Fixed it, i think the issue wasn't the write itself but the formatting.
> 
> Glad to hear that. Happy new year and thank you for keeping us posted here!

Thank you, happy new year to you as well :)