serialport/serialport-rs #243: Not a typewriter

cppcoffee, 2025-01-22:
Open the usb gadget serialport, get an error occurs.

The source code is as follows:

```
use mio_serial::{DataBits, Parity, StopBits};
use std::io::Read;
use std::io::{self, Write};
use std::time::Duration;

fn main() {
    let mut port = mio_serial::new("/dev/ttyGS0", 9600)
        .parity(Parity::None)
        .stop_bits(StopBits::One)
        .data_bits(DataBits::Eight)
        .timeout(Duration::from_secs(120))
        .open_native()
        .unwrap();       // Not a typewriter
```

Error message:

```shell
thread 'main' panicked at src/main.rs:13:10:
called `Result::unwrap()` on an `Err` value: Error { kind: Unknown, description: "Not a typewriter" }
```

sirhcel, 2025-01-22:
Sorry to hear that @cppcoffee! Let's dig into what's causing this. How do you set up the USB gadget device in question? Could you run your example with `RUST_BACKTRACE` set to `1`? For example by setting in beforehand like:

```
$ export RUST_BACKTRACE=1
```

asmforce, 2025-01-22:
It's not the same issue but seems to be related: 4.7.0 fails to work with virtual serial port created by socat utility while 4.6.1 works perfectly fine:

```shell
$ cargo build --release
   Compiling proc-macro2 v1.0.93
   Compiling unicode-ident v1.0.14
   Compiling libc v0.2.169
   Compiling pkg-config v0.3.31
   Compiling thiserror v1.0.69
   Compiling bitflags v1.3.2
   Compiling cfg-if v1.0.0
   Compiling bitflags v2.8.0
   Compiling scopeguard v1.2.0
   Compiling libudev-sys v0.1.4
   Compiling quote v1.0.38
   Compiling syn v2.0.96
   Compiling nix v0.26.4
   Compiling libudev v0.3.0
   Compiling thiserror-impl v1.0.69
   Compiling unescaper v0.1.5
   Compiling serialport v4.7.0
   Compiling experiments-rs v1.0.0 (/home/asmforce/work/vaultgroup/experiments-rs)
    Finished `release` profile [optimized] target(s) in 2.52s
$ ./target/release/experiments-rs 
Error: Custom { kind: Other, error: "Not a typewriter" }
$ cargo update serialport
    Updating crates.io index
     Locking 1 package to latest compatible version
 Downgrading serialport v4.7.0 -> v4.6.1 (available: v4.7.0)
$ cargo build --release
   Compiling serialport v4.6.1
   Compiling experiments-rs v1.0.0 (/home/asmforce/work/vaultgroup/experiments-rs)
    Finished `release` profile [optimized] target(s) in 0.52s
$ ./target/release/experiments-rs 
Written 4 bytes
Success
```

Here's the app's source code:

```rust
use std::io::{Error, Write};
use std::time::Duration;
use serialport::{FlowControl, DataBits, Parity, StopBits};

fn main() -> Result<(), Error> {
    let path = "/dev/ttyTEST";
    let builder = serialport::new(path, 9600)
        .flow_control(FlowControl::None)
        .data_bits(DataBits::Eight)
        .parity(Parity::None)
        .stop_bits(StopBits::One)
        .timeout(Duration::from_millis(1000));

    let mut port = builder.open()?;

    let data = [12u8, 23u8, 34u8, 45u8];
    match port.write(&data) {
        Ok(count) => println!("Written {} bytes", count),
        Err(e) => println!("{}", e)
    }

    match port.flush() {
        Ok(()) => println!("Success"),
        Err(e) => println!("{}", e)
    }

    Ok(())
}
```

And here is the command to create a virtual serial port:
```shell
sudo socat -d -d pty,link=/tmp/cvsimulator_sp,echo=0,perm=0777,b9600 pty,link=/dev/ttyTEST,echo=0,perm=0777,b9600
```

sirhcel, 2025-01-22:
Thank you for bringing up the issue with Linux pseudo terminals @asmforce!

At a first glance, I have no idea how to reliably detect whether the actual device is supposed to support setting control lines. But trying on best-effort looks like a way to go. See https://github.com/sirhcel/serialport-rs/commit/9b561610b08b3a4aece110b8c4ea22ecd077c862.

cppcoffee, 2025-01-23:
Testing revealed issues with `serialport` version 4.7.0, while version 4.6.1 works normally.

bachrc, 2025-01-28:
Bumping this, having the same kind of issue with version `4.7.0`, that `4.6.1` fixes.

Launched socat with this command

```bash
$ socat -d -d pty,raw,echo=0 pty,raw,echo=0
2025/01/28 10:26:13 socat[16319] N PTY is /dev/pts/1
2025/01/28 10:26:13 socat[16319] N PTY is /dev/pts/2
2025/01/28 10:26:13 socat[16319] N starting data transfer loop with FDs [5,5] and [7,7]
```

And my code is:

```rust
let mut port = serialport::new("/dev/pts/2", 115200)
        .open_native()
        .expect("Ouverture du port série");
```

Previously, with version `4.7.0`, the error was:

```
thread 'main' panicked at apps/serial_tests/src/main.rs:20:10:
Ouverture du port série: Error { kind: Unknown, description: "Not a typewriter" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I have rust in version 1.75

sirhcel, 2025-01-28:
Hello @cppcoffee and @bachrc, does https://github.com/sirhcel/serialport-rs/commit/9b561610b08b3a4aece110b8c4ea22ecd077c862 solve the issue with 4.7.0 in your case?

cppcoffee, 2025-01-31:
> Hello [@cppcoffee](https://github.com/cppcoffee) and [@bachrc](https://github.com/bachrc), does [sirhcel@9b56161](https://github.com/sirhcel/serialport-rs/commit/9b561610b08b3a4aece110b8c4ea22ecd077c862) solve the issue with 4.7.0 in your case?

The issue occurs in version 4.7.0, but version 4.6.1 does not have this problem.

sirhcel, 2025-01-31:
> In my case, the issue occurs in version 4.7.0, but version 4.6.1 does not have this problem.

Yes, i read https://github.com/serialport/serialport-rs/issues/243#issuecomment-2608646202 this way. Could you please give https://github.com/sirhcel/serialport-rs/commit/9b561610b08b3a4aece110b8c4ea22ecd077c862 a spin to check whether this also solves the issue with the USB gadget device as well?

dcallahan-qti, 2025-03-03:
I have the same issue as @cppcoffee when trying to open /dev/ttyGS0 using version 4.7.0 while 4.6.1 works.

```
serialport::new("/dev/ttyGS0", 115200)
        .data_bits(serialport::DataBits::Eight)
        .parity(serialport::Parity::None)
        .stop_bits(serialport::StopBits::One)
        .flow_control(serialport::FlowControl::None)
        .timeout(Duration::from_secs(30))
        .open()?
```

I tried sirhcel@9b56161 but I still get the same error:  

Serial(Error { kind: Unknown, description: "Invalid argument" })

sirhcel, 2025-03-03:
Thank you for trying out https://github.com/sirhcel/serialport-rs/commit/9b56161 @dcallahan-qti! Then the fix covers `socat` devices but still fails for USB serial gadgets and we need to dig deeper:

* Is `/dev/ttyG0` the gadget side of this setup? Or is this the device on the host the gadget is connected to?
* Could you please run the test binary with `strace` for a quick shot whether there is a system call which fails?
* Are you able to open `/dev/ttyG0` with another serial library like pySerial?
* Could you instrument or debug [`TTYPort::open`](https://github.com/serialport/serialport-rs/blob/960da9c41f115cf1dfb44fdbfa93c605b424c684/src/posix/tty.rs#L117) to pinpoint the cause (the `dbg` or `println` will likely suffice for instumenting)

Please feel free to ping me for help if steps are not clear to you. By the way: How does you USB gadget setup look like?

dcallahan-qti, 2025-03-04:
@sirhcel Yes, /dev/ttyGS0 is the gadget side of the setup. The MPU USB port is configured as a Device. 

I can run the tests you asked for. For now, I found if I comment out the following code in TTYPort::open, the USB Gadget "/dev/ttyGS0" opens correctly. 

```Rust
if builder.baud_rate > 0 {
    if let Some(dtr) = builder.dtr_on_open {
        port.write_data_terminal_ready(dtr)?;
    }
}
```

It looks like this was added in 82226e1.

gswebspace, 2025-03-07:
Ran into this issue and adding `preserve_dtr_on_open()` in the builder works for me for a virtual serial port created via socat.

sirhcel, 2026-03-23:
Enabling DTR on open caused quite some hassle all over the board. We therefore backtracked from this and leave it alone again since 4.7.3 (see #285).  Added USB gadget serial ports as a case for leaving DTR alone to the tracking issue https://github.com/serialport/serialport-rs/issues/292#issuecomment-4109036269 as well.