serialport/serialport-rs #246: On linux with udev, port whithout a parent should not be filtered out of `available_ports`

dzamlo, 2025-02-12:
As discussed in #245, port whithout a parent value are filtered out durin enumeration with udev 

That should likely not be the case.

sirhcel, 2025-02-14:
Thank you for opening this issue @dzamlo! Quoting from the [original discussion in #245](https://github.com/serialport/serialport-rs/discussions/245#discussion-7900385):

> I found out that the reason is that during the enumeration device without a parent are filtered out, and rfcomm (Bluetooth serial on Linux) device don't have a parent.

They seem to be the first of its kind without a parent device. Checking for the presence of a parent device has been the criterion for [quite a while](https://github.com/serialport/serialport-rs/blame/58703446cdcbf082fee50063442b41e0e3d28612/src/posix/enumerate.rs#L509).

Looking at the udev information from #245, additionally checking whether the device's sysname starts with _rfcomm_ looks like the way to go for these devices. I did a first shot on enumerating RFCOMM devices and tracing enumeration in [test/enumeration-tracing](https://github.com/sirhcel/serialport-rs/tree/test/enumeration-tracing)/[b157b7e](https://github.com/sirhcel/serialport-rs/commit/b157b7ee608c036cb229648889bdcd5a99dc3ea2). Could you please give it a spin with your RFCOMM setup?

Could you please also check what the enumeration without libudev does for RFCOMM devices? Does your device in question show up with:

```
$ export RUST_LOG=trace
$ cargo run --no-default-features --example list_ports
```

If not, is there a hint in the tracing output? And is the RFCOMM device referenced from `/sys/class/tty/`?

dzamlo, 2025-02-16:
> Could you please give it a spin with your RFCOMM setup?

The port is correctly listed. The type is Unknown and maybe should be Bluetooth.

> `cargo run --no-default-features --example list_ports`

Only my USB serial port are listed. 
the relevant tracing output is :

> 2025-02-16T12:59:48.539047Z DEBUG available_ports:path{path=Ok(DirEntry("/sys/class/tty/rfcomm0"))}: serialport::posix::enumerate: rejected cause="no device dir"

Note that most port listed with udev are also not listed without it, for either the same `no device dir` reason or for the `no port type` reason.

> additionally checking whether the device's sysname starts with rfcomm looks like the way to go for these devices.

What I don't understand is why filter device without a parent except for historical reason?

sirhcel, 2025-02-24:
> The port is correctly listed. The type is Unknown and maybe should be Bluetooth.

Great. I finally got a Bluetooth dongle and a Bluetooth serial adapter for checking this out and with [sirhcel@rfcomm-enumeration](https://github.com/sirhcel/serialport-rs/tree/rfcomm-enumeration)/https://github.com/sirhcel/serialport-rs/commit/6469fa66a40f532766d6ce5dc3d52fd54e52bfa9 there is an updated variant of the enumeration available which reports RFCOMM devices as `BluetoothPort`. With and without libudev. Could you please give it a spin?

```
$ cargo run --example list_ports
$ cargo run --no-default-features --example list_ports
```

Both enumeration variants should produce the same output.

> > `cargo run --no-default-features --example list_ports`
> 
> Only my USB serial port are listed. 

Are there serial ports besides RFCOMM missing?

> Note that most port listed with udev are also not listed without it, for either the same `no device dir` reason or for the `no port type` reason.

I'm seeing exactly the same output from `list_ports` with or without libudev. Which ports are you missing? What kind of machine are you using? Is this more of an embedded or desktop setup?

> What I don't understand is why filter device without a parent except for historical reason?

Me neither. But this criterion is there "forever" and filters out a whole bunch of TTY devices which aren't actual serial ports on my machine.

sirhcel, 2025-02-24:
My Bluetooth serial setup is a TP Link UB500 dongle and a HC-05 Bluetooth serial module. The latter runs at its default speed of 9,600 Baud.

When running the tests from serialport, there are errors from [clearing the input/output buffers](https://github.com/sirhcel/serialport-rs/blob/6469fa66a40f532766d6ce5dc3d52fd54e52bfa9/tests/test_baudrate.rs#L32) of the RFCOMM device. They appear spuriously and it looks like the underlying `tcflush` returns -1 in case they fail.

There are more issues related to baud rate changes which I'm ignoring in the first place.

Does Bluetooth serial communication work reliably in your case? Which hardware are you using? Are you able to run serialport's test in your setup? Like:

```
$ export RUST_BACKTRACE=1
$ export SERIALPORT_TEST_PORT_1 /dev/rfcomm0
$ export SERIALPORT_TEST_PORT_2 /dev/ttyUSB0
$ cargo test --test test_baudrate
```

dzamlo, 2025-03-03:
The ports missing without udev are tty starting with `/dev/ttyS`. In retrospect, I believe they should not be listed as they are not actual serial ports.

> Does Bluetooth serial communication work reliably in your case? Which hardware are you using? Are you able to run serialport's test in your setup? Like:

I'm using an ESP32 using the BluetoothSerial > SerialtoSerialBT Arduino example.

The communication works reliably and the `test_baudrate`test run successfully.

It seems that the baud rate on the Bluetooth serial connection is ignored.

sirhcel, 2025-03-21:
> The ports missing without udev are tty starting with `/dev/ttyS`. In retrospect, I believe they should not be listed as they are not actual serial ports.

Than we are on the same page here. Good.

> The communication works reliably and the `test_baudrate`test run successfully.
> 
> It seems that the baud rate on the Bluetooth serial connection is ignored.

Good to hear that. As far as I'm overlooking things here, this is something where you have to use a different mechanism for the "far end" of the Bluetooth serial connection.

> EDIT: the test_timeout test fails: https://gist.github.com/dzamlo/b8d6864203c27ebf28a2bc1f8193de28

Thanks for the test log! This is not great but the changes made with https://github.com/sirhcel/serialport-rs/commit/6469fa66a40f532766d6ce5dc3d52fd54e52bfa9 juts address the enumeration and I don't see how they influence timeout behavior in the first place.

From my point of view this issue was already present before and does not prevent merging the changes for the enumeration for me.

Do you see any reason for not integrating the changes for enumeration RFCOMM devices at all?

PR #255 (2025-03-25): Enumerate Bluetooth serial (RFCOMM) devices on Linux too
* RFCOMM devices will be enumerated now (see #246)
* There are still issues with timeouts, ...
    * But they are independent from the enumeration
    * And some of them like baud rate setting are likely related to the nature of this connection type
* This PR also drive-by adds a test case for baud rates
* Fixes #246
closed 2025-03-25 by eldruin