serialport/serialport-rs #247: "unescaper" dependency is unnecessary when libudev is not used

JonathonReinhart, 2025-02-13:
The `unescaper` crate is only used in one function, [`udev_property_encoded_or_replaced_as_string()`](https://github.com/serialport/serialport-rs/blob/58703446cdcbf082fee50063442b41e0e3d28612/src/posix/enumerate.rs#L88-L98), which is guarded by the `libudev` feature.

However, the `unesecaper` dependency is not conditional on the `libudev` feature: [Cargo.toml](https://github.com/serialport/serialport-rs/blob/58703446cdcbf082fee50063442b41e0e3d28612/Cargo.toml#L22). It is included for all linux builds.

sirhcel, 2025-02-14:
Thank you @JonathonReinhart! Good catch.

To my understanding, enabling an optional dependency based on a feature of the crate is possible with the `"dep:[...]"` syntax since Rust 1.60 but not with our current MSRV 1.59. Do you know whether there is a way to get this working with 1.59?

What's the actual hassle you are experiencing with having this unused dependency on board?

JonathonReinhart, 2025-02-17:
> Do you know whether there is a way to get this working with 1.59?

I'm not sure, I'm not that familiar with Cargo's dependency features.

> What's the actual hassle you are experiencing with having this unused dependency on board?

I'm considering importing the `serialport` crate into Android's vendored repository [`android-crates-io`](https://android.googlesource.com/platform/external/rust/android-crates-io/), which would also require importing the `unescaper` dependency. It's not a showstopper, just an additional chore.

VMshall, 2026-04-05:
Hii! there , i'm willing to give it a shot