serialport/serialport-rs #248: "cargo test" fails and TESTING.md provides no guidance

JonathonReinhart, 2025-02-25:
[TESTING.md](https://github.com/serialport/serialport-rs/blob/main/TESTING.md) says:

> How to test `serialport-rs` for development.
>
> Without hardware:
>
>  1. Compilation
>  2. `cargo test`

---

When I try that it fails:

```
$ cargo test --no-default-features                                                                                                                                                           
    Finished `test` profile [unoptimized + debuginfo] target(s) in 0.06s
     Running unittests src/lib.rs (target/debug/deps/serialport-083848d9b1c403f1)
                                                                                                                       
running 6 tests                                                                         
test posix::poll::tests::clamped_millis_c_int_is_monotonic ... ok                       
test posix::poll::tests::clamped_millis_c_int_zero_is_zero ... ok                       
test posix::poll::tests::clamped_time_spec_is_monotonic ... ok                        
test posix::poll::tests::clamped_time_spec_zero_is_zero ... ok                        
test posix::tty::test_ttyport_into_raw_fd ... ok                                      
test tests::timeout::tests::basic_durations_properties ... ok
                                                                                                                       
test result: ok. 6 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.00s

     Running tests/config.rs (target/debug/deps/config-ae02c05788e27e89)

running 0 tests                                                                       
                                                                                                                       
test result: ok. 0 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.00s
                                                                                                                       
     Running tests/test_baudrate.rs (target/debug/deps/test_baudrate-b7c732862a686999)
                                                                                                                       
running 27 tests                                                                                                 
test builder::test_non_standard_baud_rate_fails_where_not_supported::case_1 ... ignored      
test builder::test_non_standard_baud_rate_fails_where_not_supported::case_2 ... ignored
test builder::test_non_standard_baud_rate_fails_where_not_supported::case_3 ... ignored
test builder::test_non_standard_baud_rate_succeeds_where_supported::case_1 ... FAILED                            
test builder::test_non_standard_baud_rate_succeeds_where_supported::case_2 ... FAILED        
test builder::test_non_standard_baud_rate_succeeds_where_supported::case_3 ... FAILED
test builder::test_standard_baud_rate::case_1 ... FAILED 
test builder::test_standard_baud_rate::case_2 ... FAILED                            
test builder::test_standard_baud_rate::case_3 ... FAILED                                     
test new::test_non_standard_baud_rate_fails_where_not_supported::case_1 ... ignored
test new::test_non_standard_baud_rate_fails_where_not_supported::case_2 ... ignored
test new::test_non_standard_baud_rate_fails_where_not_supported::case_3 ... ignored 
test new::test_non_standard_baud_rate_succeeds_where_supported::case_1 ... FAILED            
test new::test_non_standard_baud_rate_succeeds_where_supported::case_2 ... FAILED
test new::test_non_standard_baud_rate_succeeds_where_supported::case_3 ... FAILED
test new::test_standard_baud_rate::case_1 ... FAILED                                
test new::test_standard_baud_rate::case_2 ... FAILED                                         
test new::test_standard_baud_rate::case_3 ... FAILED
test set_baud::test_non_standard_baud_rate_fails_where_not_supported::case_1 ... ignored
test set_baud::test_non_standard_baud_rate_fails_where_not_supported::case_2 ... ignored                     
test set_baud::test_non_standard_baud_rate_fails_where_not_supported::case_3 ... ignored     
test set_baud::test_non_standard_baud_rate_succeeds_where_supported::case_1 ... FAILED
test set_baud::test_non_standard_baud_rate_succeeds_where_supported::case_2 ... FAILED
test set_baud::test_non_standard_baud_rate_succeeds_where_supported::case_3 ... FAILED                       
test set_baud::test_standard_baud_rate::case_1 ... FAILED                                    
test set_baud::test_standard_baud_rate::case_2 ... FAILED
test set_baud::test_standard_baud_rate::case_3 ... FAILED                         
                                                                
failures:

---- builder::test_non_standard_baud_rate_succeeds_where_supported::case_1 stdout ----
thread 'builder::test_non_standard_baud_rate_succeeds_where_supported::case_1' panicked at tests/config.rs:24:37:
called `Result::unwrap()` on an `Err` value: EnvVarMissing { name: "SERIALPORT_TEST_PORT_1" }
[... 69 lines cut ...]

failures:
    builder::test_non_standard_baud_rate_succeeds_where_supported::case_1
    builder::test_non_standard_baud_rate_succeeds_where_supported::case_2
    builder::test_non_standard_baud_rate_succeeds_where_supported::case_3
    builder::test_standard_baud_rate::case_1
    builder::test_standard_baud_rate::case_2
    builder::test_standard_baud_rate::case_3
    new::test_non_standard_baud_rate_succeeds_where_supported::case_1
    new::test_non_standard_baud_rate_succeeds_where_supported::case_2
    new::test_non_standard_baud_rate_succeeds_where_supported::case_3
    new::test_standard_baud_rate::case_1
    new::test_standard_baud_rate::case_2
    new::test_standard_baud_rate::case_3
    set_baud::test_non_standard_baud_rate_succeeds_where_supported::case_1
    set_baud::test_non_standard_baud_rate_succeeds_where_supported::case_2
    set_baud::test_non_standard_baud_rate_succeeds_where_supported::case_3
    set_baud::test_standard_baud_rate::case_1
    set_baud::test_standard_baud_rate::case_2
    set_baud::test_standard_baud_rate::case_3

test result: FAILED. 0 passed; 18 failed; 9 ignored; 0 measured; 0 filtered out; finished in 0.00s

error: test failed, to rerun pass `--test test_baudrate`
```

sirhcel, 2025-02-25:
Good catch @JonathonReinhart! Well ... `TESTING.md` is outdated and we should update it. 😅

Meanwhile: To run the tests not requiring serial devices, you can invoke them like:

```
$ cargo test --features ignore-hardware-tests
```

Or to run them with two serial devices connected:

```
$ export SERIALPORT_TEST_PORT_1=$(realpath /dev/ttyX)
$ export SERIALPORT_TEST_PORT_1=$(realpath /dev/ttyY)
$ cargo test
```

sirhcel, 2025-03-09:
`TESTING.md` updated with #249.

closed 2025-03-09 by sirhcel