serialport/serialport-rs #250: Reading individual USB packets.

elpiel, 2025-03-09:
USB data packets are usually up to 64 bytes and I would like to be able to read those packets instead of reading just raw bytes when they come (Read).

This could greatly improve building protocols for embedded firmware as you can use the same exact encoding on the device side and on the host side.

sirhcel, 2025-03-09:
This is definitely something useful when it comes to communicating with embedded devices. Could you flesh out your idea and the goal bit more?

Maybe there are already tailored solution to this readily available: For example [postcard-rpc](https://github.com/jamesmunns/postcard-rpc) which supports raw USB transport and could be even used on top of existing USB serial protocols without interfering directly with the latter. And it allows to share the exact same protocol/interface definition code between the devices communicating.

The serialport library operates on top of the device drivers which abstract the USB communication with the actual device away. Directly speaking to a USB serial device from the serialport crate is effectively incorporating a userspace driver for a certain USB serial protocol into the crate.

elpiel, 2025-03-09:
My idea was simple: Instead of providing only Read/Write to be able to read a USB frame through methods.
I took a look at postcard-rpc, at least on the host side, it does a buffering loop to make sure and read bytes but process 0, 1 or more messages.
I need to read a bit more and understand the rpc part of it though. Underneath it uses nusb which is low-level usb crate I've taken a look at.

sirhcel, 2025-03-09:
> My idea was simple: Instead of providing only Read/Write to be able to read a USB frame through methods.

Do you know more about how to achieve this? I'm asking because I don't know of an interface where serialport could get/inspect the USB traffic from a USB serial device. The device driver abstracts this completely away an just exposes data and device control though the serial device serialport-rs operates on.

elpiel, 2025-03-10:
As far as I saw, it can't. I thought that serialport uses something like `nusb` to operate on the usb device and not the OS io.

I took a look at postcard-rpc and indeed there, at least for the cobs encoder/decoder, the crate itself allows you to find the frames' ends and decode each (see https://docs.rs/cobs/latest/cobs/fn.decode_in_place.html). I'm still looking around to get a general feeling this can be achieved (in serialport or not).
`nusb` has methods for bulk in/out but it also has a much lower-level api and I need to get a bit more familiar with the usb protocol.

I'm open to ideas too.

sirhcel, 2025-03-21:
> As far as I saw, it can't. I thought that serialport uses something like `nusb` to operate on the usb device and not the OS io.

No, we are taking the cozy approach here and are just abstracting over the OS interface which in turn uses its OS specific device drivers.

> I took a look at postcard-rpc and indeed there, at least for the cobs encoder/decoder, the crate itself allows you to find the frames' ends and decode each (see [...]

How are things going here? Did you find something suitable for your needs?

> `nusb` has methods for bulk in/out but it also has a much lower-level api and I need to get a bit more familiar with the usb protocol.

While USB serial protocols don't look super complex, implementing some userspace drivers in Rust looks somewhat like a mighty thing to do in the historic context of this crate. This would require implementing at least some of most-common USB serial protocols and the device enumeration. This might be great fun to play with - but looking at the current backlog here, this is nothing I will do in the short term. And on top, this also needs maintainership in the long run.