serialport/serialport-rs #251: v4.7. (set DTR on opening) breacks some window - arduino application

NWPoul, 2025-03-16:
setting by default,
dtr_on_open: Some(true)

can (and in my case do)) break communication to arduino on windows platform

As far as i anderstood fighting with sudden bug in my app after update
arduino in some (IMHO default) configuration whatch that signal to reset upon port opening

IMHO this options should not be set as default, but revert it may be again a braking change (relative to 4.7 version) at least it should be waning some where in change log or read me

NWPoul, 2025-03-22:
Hellj @sirhcel !
Pls don't be sorry and thank You for the crate!

In my understanding the issue cased wierd Arduino's using DTR signal for initialisation reset/bootload seq on port opening

[https://forum.arduino.cc/t/soft-reset-on-arduino-through-starting-serial-comm/26755](url)

This reset takes something from 800 to 3000 ms so IMHO the real solution which works in https://github.com/serialport/serialport-rs/issues/204#issuecomment-2246398222 is delay in
`std::thread::sleep(Duration::from_millis(1500));`
not the DTR settings

I think tha DTR_on_open option is indeed usefull (and can solve other issues) but should't be set as default
but as I said,? revert it can again became breaking change btw crate vertions so IMHO solving this bug/issue can be just somehow obvious varning somewhere)

vonPB, 2025-07-31:
Just wanted to mention that I am glad I found this issue.
I spent a few hours wondering why my app stopped receiving any bytes from the arduino.
Setting `dtr_on_open(false)` fixed the problem for me.

Many thanks!