serialport/serialport-rs #252: Potential Regression in sending raw bytes over serial (from 4.6.1 to 4.7.0)

DerAndereJohannes, 2025-03-21:
Dear Maintainers,

I have created an ESP32 application using ESP-IDF to convert raw bytes that I send over 8 data pins to an LPT port. I have tested that this program works perfectly with the following python code:

```python3
import serial
import time

ser = serial.Serial('COM6', 115200, timeout=1)

for b in range(50, 256):
    ser.write(bytes([b]))
    ser.flush()
    print(f"Sent: 0x{b:02X}, {b}")
    time.sleep(0.2)
```

However, when I try to run it in rust using serialport, it also works perfectly, except for the 7th bit not working. This means that any value involving 64 in the computation does not work. This seems to be the case on version `4.7`. When I return to `serialport = "=4.6.1"`, the 7th bit does end up working (with the same code) and everything runs like the python version. Here is my rust code:

```rust
use std::{thread::sleep, time::Duration};

fn main() {
    let port_name = "COM6"; // Adjust for your setup
    let baud_rate = 115200;

    let mut port = serialport::new(port_name, baud_rate)
        .timeout(Duration::from_secs(1))
        .data_bits(serialport::DataBits::Eight)
        .stop_bits(serialport::StopBits::One)
        .parity(serialport::Parity::None)
        .flow_control(serialport::FlowControl::None)
        .open()
        .expect("Failed to open serial port");

    let tosend: Vec<u8> = Vec::from_iter(50..=255);

    for b in tosend {
        port.write(&[b]).expect("Failed to write byte");
        port.flush().unwrap();

        sleep(Duration::from_millis(200));
    }
}
```

Perhaps I am misunderstanding something about my code and the new serialport version, but I believe that this case could be interesting to you.

Rust Version: 1.85.0
Operating System: Windows 10 22H2 19045.5608

sirhcel, 2025-03-21:
Thank you for this report @DerAndereJohannes! I ran the Rust test application from above (as [`issue_252_send_data.rs`](https://github.com/sirhcel/serialport-rs/blob/771e0ab71531e47e7007dae73fd82134990d2e62/examples/issue_252_send_data.rs)) on both, 4.6.1 and 4.7.0 on Windows 10. I got identical and the expected output in both cases: a sequence of the numbers 50 to 255.

> However, when I try to run it in rust using serialport, it also works perfectly, except for the 7th bit not working.

Could you please elaborate a bit more which output you are expecting and what you are actually getting in your setup? At which places in your setup are you making your observations?

I'm using a CP2102N, recording the output from the application with a logic analyzer, and finally checking the sequence of bytes with a spreadsheet. Everything looks hunky dory so far in my setup.

> This means that any value involving 64 in the computation does not work. This seems to be the case on version `4.7`. When I return to `serialport = "=4.6.1"`, the 7th bit does end up working (with the same code) and everything runs like the python version.

Which computation are you referring to here? The application is just outputting the sequence from `tosend`. Is this something happening on the ESP32? And wouldn't bit 7 be 2^7 == (1 << 7) == 128?

DerAndereJohannes, 2025-03-24:
Thank you so much for the swift response. I also understand how bizarre my situation is. 

My setup is basically: PC SERIAL -> ESP32-WROOM-32 -> 8 gpio pins into a BIOPAC MP-150 digital inputs (one bit per channel) and yes I am using bitshifts on the esp to set the channels. Here is the code for the ESP:

```c
uint32_t bitmask = 0;
for (int i = 0; i < ARRAY_SIZE(pins1); i++) {
    if (value & (1 << i)) {
        bitmask |= (1 << (pins1[i]));
    }
}

REG_WRITE(GPIO_OUT_W1TS_REG, bitmask);
vTaskDelay(PIN_RESET_TIME_MS);
REG_WRITE(GPIO_OUT_W1TC_REG, bitmask);
```

Here is the image I get on the Acqknowledge software (the software to record data from the BIOPAC) for serialport v4.6.1:

[image]

Here everything is fine (the bits just go up sequentially. The first combination containing the 7th bit being 192. (This is the output I am expecting)

Now here is what I get with v4.7 (funnily enough I do see it sometimes now which confuses me more):

[image]

I hope that this information helps a bit more.

sirhcel, 2025-03-24:
You are welcome @DerAndereJohannes! Thank you for the update. Every bizarre-looking situation I encountered turned out to have a natural and explainable cause so far. I bet it's the same here in the end.

The code transforming the byte into the GPIO bit pattern looks unsuspicious in the first place. I see the different patterns is the screenshots but I still don't have an explanation just from snippets above.

What about checking things in a more isolated fashion next? Could you please run a modified firmware on the ESP32 which just echoes back the received characters and read them back on the host? Could you simulate the received data on the ESP32 and check if this still gives the expected output?

Are you using just the ESP32-WROOM-32 in a piece of custom hardware or a ESP32-DevKitC just wired to the BIOPAC inputs?