serialport/serialport-rs #253: Bad serial number

RichardWGNR, 2025-03-24:
### Problem Description / Reproduction

I use `serialport::available_ports()` to get information about the current ports.
I have the next code:

```rust
        let ports = serialport::available_ports().unwrap();
        for port in ports {
            match port.port_type {
                SerialPortType::UsbPort(ref device) => {
                    debug!("S/N: {:?}", device.serial_number);
                },
                _ => {},
            }
        }
```

Which outputs the following to the command line:
```
DEBUG ThreadId(12) tmcs_client::flash::resolver::usb_resolver: S/N: Some("TMCS")
```

But the serial number of my device is **not “TMCS”** at all, but “TMCS_B000000000”.

[image]

[image]

### Expected

Correct serial number

### Actual

Truncated serial number

### My environment

| Package    | Version |
|------------|---------|
| Windows    | 11      |
| Rust       | 1.85.1  |
| serialport | 4.7.0   |

sirhcel, 2025-03-24:
Good catch @RichardWGNR! Sorry, that you ran into it.

When replacing the former regex used there with bespoke matching, at least the underscore went missing from the practically relevant characters. #254 sets out to bring it back.

closed 2025-03-25 by eldruin

RichardWGNR, 2025-03-25:
> Good catch [@RichardWGNR](https://github.com/RichardWGNR)! Sorry, that you ran into it.
>
> When replacing the former regex used there with bespoke matching, at least the underscore went missing from [...]

Thank you very much!

PR #254 (2025-03-25): Fix parsing serial numbers with underscores from Windows HWID
* Fixes #253
* We've forgotten to at least include the underscore when replacing the former regex