serialport/serialport-rs #257: No com port number

RichardWGNR, 2025-03-27:
### Problem Description

I use `serialport::available_ports()` to get information about the current ports.
I have the next code:

```rust
        let ports = serialport::available_ports().unwrap();
        for port in ports {
            debug!("{port:?}");
        }
```

Which outputs the following to the command line:

```
18:56:39 [windows] 2025-03-26T13:56:39.768815Z DEBUG ThreadId(12) tmcs_client::flash::resolver::usb_resolver: [
18:56:39 [windows]     SerialPortInfo {                                                                                                                                                                                                                                                                             
18:56:39 [windows]         port_name: "COM1",                                                                                                                                                                                                                                                                       
18:56:39 [windows]         port_type: Unknown,                                                                                                                                                                                                                                                                      
18:56:39 [windows]     },                                                                                                                                                                                                                                                                                           
18:56:39 [windows]     SerialPortInfo {                                                                                                                                                                                                                                                                             
18:56:39 [windows]         port_name: "",  // ATTENTION HERE                                                                                                                                                                                                                                                                         
18:56:39 [windows]         port_type: UsbPort(                                                                                                                                                                                                                                                                      
18:56:39 [windows]             UsbPortInfo {                                                                                                                                                                                                                                                                        
18:56:39 [windows]                 vid: 1155,                                                                                                                                                                                                                                                                       
18:56:39 [windows]                 pid: 22336,                                                                                                                                                                                                                                                                      
18:56:39 [windows]                 serial_number: Some(                                                                                                                                                                                                                                                             
18:56:39 [windows]                     "TMCS_B000000000",                                                                                                                                                                                                                                                           
18:56:39 [windows]                 ),                                                                                                                                                                                                                                                                               
18:56:39 [windows]                 manufacturer: Some(                                                                                                                                                                                                                                                              
18:56:39 [windows]                     "STMicroelectronics.",                                                                                                                                                                                                                                                       
18:56:39 [windows]                 ),                                                                                                                                                                                                                                                                               
18:56:39 [windows]                 product: Some(                                                                                                                                                                                                                                                                   
18:56:39 [windows]                     "TMCarSolutions: Common Bootloader (COM 10)",                                                                                                                                                                                                                                
18:56:39 [windows]                 ),                                                                                                                                                                                                                                                                               
18:56:39 [windows]             },                                                                                                                                                                                                                                                                                   
18:56:39 [windows]         ),                                                                                                                                                                                                                                                                                       
18:56:39 [windows]     },                                                                                                                                                                                                                                                                                           
18:56:39 [windows] ]   
```

As you and I can see, there is no port name for the second device.

**I don't know what the problem is, and I don't know how to reproduce this on another computer.**

Other programs easily recognize the COM port name of this device.

**Serial** Debug Assistant

[image]

**Windows registry**

[image]

### My environment

| Package    | Version |
|------------|---------|
| Windows    | 11      |
| Rust       | 1.85.1  |
| serialport | 4.7.0   |

RichardWGNR, 2025-03-27:
So far, I've done a workaround using the following function to which I give a string with the FriendlyName of the device.

```rust
    fn guess_com_port_from_str<S>(str: S) -> Option<String>
    where
        S: AsRef<str>
    {
        static RGX: LazyLock<Regex> = LazyLock::new(|| {
            Regex::new(r#"\(COM\s*(?<num>\d+)\s*\)"#).unwrap()
        });

        if let Some(caps) = RGX.captures(str.as_ref()) {
            let port_num = caps.name("num").unwrap().as_str();
            return Some(format!("COM{port_num}"))
        }

        None
    }
```

```rust
list.insert(unique_device_sequence, UsbDevice {
                sw_name: info.product_string().map(|v| v.to_owned()),
                serial_number: sn.to_owned(),
                com_port: if device.port_name.len() > 0 {
                    device.port_name
                } else {
                    let com_port = usb_port.product
                        .as_ref()
                        .and_then(|product| Self::guess_com_port_from_str(product));

                    if let Some(port) = com_port {
                       port
                    } else {
                        continue;
                    }
                },
            });
```

sirhcel, 2025-08-25:
I'm sorry that you are experiencing this strange error. Does it still exist?

RichardWGNR, 2026-03-17:
> Do you still see this issue [@RichardWGNR](https://github.com/RichardWGNR)? If you are, which device driver handles the device you are getting no name for? I case you are still interested, instrumenting the enumeration - especially `PortDevice::name` - would be my next step and I can help you with that.

Thank you! I haven't used USB in quite a while, so I can't test anything right now.

I used the Virtual Com Port driver from STMicroelectronics.
I simply changed the FriendlyName in the Registry Editor, after which the port name stopped appearing in your library.

Changing the FriendlyName is necessary to customize the names of my devices for free. Because currently, this is only possible using your own driver and a VID/PID license.