serialport/serialport-rs #265: Use flock for exclusive serial port access on Linux

Tiwalun, 2025-05-13:
This is a follow-up to #26, it seems that only `tiocexcl` is currently used, but not `flock`.

It would be nice to get support for `flock` as well, to catch more cases in which another application is already using the serial port.

I can create a PR adding support for this, if it something that would be accepted.

sirhcel, 2025-05-13:
> It would be nice to get support for `flock` as well, to catch more cases in which another application is already using the serial port.

Thank you for raising this issue! I have not looked into the different locking mechanisms so far. What would be the benefit of using `flock` over `tiocexcl`?

> I can create a PR adding support for this, if it something that would be accepted.

If you care for it - I bet others will as well. This would be a good reason for further discussion and seriously looking into a potential PR.

Tiwalun, 2025-05-14:
I've opened a PR with an implementation.

In short, the issue this prevents is when e.g. picocom is opened first, and then we try to open a port using this crate. picocom doesn't seem to use `tioexcl`, so nothing prevents us from calling `open` again.

The `flock` call is in addition to `tioexcl`, it doesn't replace it. `tioexcl` is still used to prevent other application from using the port after we've opened it.

closed 2025-09-30 by eldruin